texhex/BiosSledgehammer #6: Support two step BIOS update process?

texhex, 2016-11-07:
Some devices require a two-step process to update. For example the [HP Compaq Pro 6300 BIOS 3.03](http://h20564.www2.hp.com/hpsc/swd/public/detail?sp4ts.oid=5232700&swItemId=vc_176829_1&swEnvOid=4158#tab-history) requires the user first to update to 2.99 (if current version < 3.00) and then install 3.04. If trying to flash directly from anything below 2.99 to 3.xxx, the update fails. 

Given that only this device seems to behave like this, it's questionable if this should be supported.

niks17, 2017-01-30:
Hi i have HP Elitebook 850 with locked bios, and am trying this your thing, but i dont know how to start

texhex, 2017-01-30:
Sorry, but I think this project is not for you. This tool is for mass updates of devices, not a single one. Also, what does this have to do with a two step BIOS update processs?

texhex, 2017-10-31:
Given that nobody has commented to this in a year, this option doesn't seem to be required. 

Issue will be closed.

karlis-vitols, 2018-07-19:
Hello.
I wonder if I can reopen this?
The 6300/8300 line (so SFF and various tower variants) were directly preceding the 600/800 G1, thus were pretty common. I, for example, have around 300 of those. HP SSM is useless in this case, as it also cannot do this version jump, nor it will attempt to install anything at all for those desktops when running on W10, as there are no drivers for them (officially only supports 8.1 at most, while, obviously, runs W10 perfectly fine).
Thus there are two options - either separate TS steps with WMI filtering just for those desktops, or your magic via Sledgehammer. 
I do, however, understand the annoyance of this specific problem.

texhex, 2018-07-19:
I just reopened the issue so it’s visible again, so the answer is yes :-). 

Implementing this feature however will most likely not happen simply because you are only the second person to react on this, hence it doesn’t seem to be a common problem. Also, this feature would benefit only one rather old series but would make the code more complicated for everyone.

What I could offer is that I present a solution I came up for this problem some time ago. This was never put to the test, because the company decided to simply replace all 6300 devices, but I’m confident that it works. 

If you are willing to test it, we can discuss here the details and if works as expected, I would create a section in README.MD to explain this in full detail. But I would need the feedback from you at the end if it worked as planned or not.

datagutten, 2018-07-19:
Could it be made as a separate batch or PS script which can be called as a command in BIOS-Update.txt? That way these models can be supported without changing the main code.
The company I work for have about 40 8300 SFF, but I am not sure if they will get Windows 10 or get replaced.

texhex, 2018-07-19:
The general procedure I had prepared was like this:

-	A new BIOS Sledgehammer folder is created, completely separated from the “main” folder. 
-	Our main folder is `\Scripts\BiosSledgehammer` so the new folder was called `\Scripts\BiosSledeghammer_PreUpdate`.
-	To that new folder, copy all files from the main BIOS Sledgehammer folder
-	**Delete** everything in `BiosSledeghammer_PreUpdate\Shared` and in `BiosSledeghammer_PreUpdate\Models` except the folder `HP Compaq Pro 6300`
-	Inside that folder delete all folders and files except `FIRST-FLASH-BIOS-2.99` and `BIOS-Update.txt`
-	Rename the folder First-Flash.. to `BIOS-2.99`
-	Edit `BIOS-Update.txt` and change the VERSION parameter to `VERSION == 2.99`

This of course requires that the folder `BIOS-2.99` contains the BIOS update files. If your installation does not, just download the [latest release]( https://github.com/texhex/BiosSledgehammer/releases/), unpack it and run `StartExampleDownloads.bat` which will download all files from HP.com.

Next, head over to MDT/SCCM and search the section where you have your four to five calls to BIOS Sledgehammer normally. Just **before** that first call, insert a new section and set a WMI filter to it so this section will only run when the computer/device model is *HP Compaq Pro 6300*. Insert a new command and let SCCM/MDT execute `cmd.exe /c "%SCRIPTROOT%\BiosSledgehammer_PreUpdate\RunVisible.bat"`. Add a restart command after that.  If you want to be sure, add another command and restart, duplicating the one you already created. This is not necessary, just a precaution.

What will happen is the following:

* For any device that is **NOT** a 6300, nothing will change because the section is ignored as the WMI filter does not match. This means, only the main BIOS Sledgehammer script is run as before.
* In case the device is a Pro 6300, `\BiosSledeghammer_PreUpdate` will run and update the BIOS to 2.99. 
* After the restart(s), the task sequence will continue, starting the main BIOS Sledgehammer (`\BiosSledeghammer`) and updating the BIOS to 3.x (or anything newer) which is now possible because the device has BIOS 2.99 already
* If an already updated (BIOS 3.x) Pro 6300 is started again, the `\BiosSledeghammer_PreUpdate` will be executed, but since the BIOS is already newer, it won’t update anything

As this `\BiosSledeghammer_PreUpdate` folder is only used for one model, it’s not necessary to keep it up to date with new releases or files, since it only serves one purpose and it’s highly unlikely that HP will ever update the BIOS 2.99 release. Hence, this won’t introduce any additional maintenance effort. 

A word of warning: We had some problems with a few ProDesk G1 devices that were not able to read our “normal” password files, they just insisted that the password is wrong (which it wasn’t). 

If this happens, recreate the password files with `\HPQFlash\HpqPswd.exe` from BIOS-2.99 and store them in `BiosSledeghammer_PreUpdate\PwdFiles`. It seems that HP uses a different file structure, so the 2.99 BIOS update might not be able to read password file correctly. 

I hope I didn't miss anything because all this is already deleted (as it was never used) so I tried to describe it as I remember. I would appreciate if this could be put to the test.

karlis-vitols, 2018-07-23:
Hi.
Thank you, this should work without a hitch, I think. I will be able to test within 1-2 weeks and report back, if the case can be kept open that long?

texhex, 2018-07-23:
Keeping this issue open is not a problem as this is just a process topic and not related to the upcoming 5.0 BETA release. 

Take you time and let me know the outcome. If it does not work, please include the verbose log so I can have a look.

texhex, 2018-08-07:
Thanks for the check and report, I'm working on the update of README.MD as suggested.

texhex, 2018-08-07:
README.MD now contains a section where I tried to describe this: https://github.com/texhex/BiosSledgehammer#two-step-bios-update-process

Most is copy & pasted from here, but it would be nice if you could check this section and tell me if me if I missed something. 

If it's OK, I will close this issue.

karlis-vitols, 2018-08-08:
Seems that all steps are included, so we can close this.
Thank you!

texhex, 2018-08-08:
You're welcome. Thanks for checking, will close issue now.

closed 2018-08-08 by texhex