texhex/BiosSledgehammer #12: Rework StartExampleDownloads.ps1

texhex, 2017-05-18:
Right now, this scripts downloads and processed every SPDownload.txt file separately. As we now often have the same download file (TPM, ME) for several models, we should try to download the files only once to save bandwidth and time.

texhex, 2017-05-18:
And we should change the name of the script as well...

napzero, 2018-03-11:
How would you account for different models using the same download? Would you separate the downloads from the models by having a separate softpaq folder, and having the model configurations point to the softpaqs?

texhex, 2018-05-10:
@napzero: The basic idea was that StartExampleDownloads.ps1 would first scan for any ``SPDownlod.txt`` and create an internal list that includes the download URL and the target path. Then, all duplicate URL would be removed and the path would be "attached" to already existing download URL. 

That way, a URL that is used by more than one model would only be downloaded once and then copied to all folders that used that URL. That would have been especially useful for the TPM update downloads as we have more than 10 models that all used the TPM update.

However, since we have the \Shared folder from v4 onwards, there are very few duplicate downlaods. For example, the TPM update is now stored only once inside the \Shared\TPM SLB 9670 folder so it's also downloaded only once. The same applies to BIOS updates for model families as EliteBook 830/840/850 and other files. 

I will close this issue.