texhex/BiosSledgehammer #15: Intel ME Firmware Fails

bandwith, 2017-06-06:
Hangs when attempting to install ME Firmware if ME Device Driver is not found. 

results from manual attempt
```C:\Users\<user>\AppData\Local\Temp\ME-9.5.61.3012\FW64>.\update64.bat

C:\Users\<user>\AppData\Local\Temp\ME-9.5.61.3012\FW64>FWUpdLcl64.exe -f ../ME/ME9.5_5M_Production.bin -allowsv -y

Intel (R) Firmware Update Utility Version: 9.5.61.3012
Copyright (C) 2007 - 2017, Intel Corporation.  All rights reserved.

Error 8193: Intel (R) ME Interface : Cannot locate ME device driver

C:\Users\<user>\AppData\Local\Temp\ME-9.5.61.3012\FW64>xcopy "MEFWDetailFile.exe" "C:\Program Files\Intel\Intel(R) Management Engine Components\" /y
C:MEFWDetailFile.exe
1 File(s) copied

```

Truncated output from BiosSledgehammer.ps1 -Verbose

```***** ::END:: Discovery Tool Output *****************************
Processing XML output (C:\Users\<user>\AppData\Local\Temp\*_System_Summary.xml)...
Reading finished
Management Engine information:

  Firmware Version ..: 9.5.15.1730
  Feature Level .....: Consumer
  Provisioned .......: Not Provisioned

Current ME firmware version: 9.5.15.1730  (Numerical: 9.5.15.1730)
Desired ME firmware version: 9.5.61.3012  (Numerical: 9.5.61.3012)
  ME update required!
The ME update will take some time, please be patient.
VERBOSE: Invoke-UpdateProgram() started
Checking if system is on AC or DC (battery) power...
VERBOSE: Perform operation 'Enumerate CimInstances' with following parameters, ''namespaceName' = root\cimv2,'className' = Win32_Battery'.
VERBOSE: Operation 'Enumerate CimInstances' complete.
  Battery status is OK or system has no battery.
::: Preparing launch of update executable :::
Copy from [\\<server>\e$\DeploymentShare\Scripts\custom\BiosSledgehammer\Models\HP ZBook 14\ME-9.5.61.3012\]
       to [C:\Users\<user>\AppData\Local\Temp\ME-9.5.61.3012\] ...
VERBOSE: Copy done
Trying to delete [*.log] in target folder
Starting:
  C:\Users\<user>\AppData\Local\Temp\ME-9.5.61.3012\CallInst.exe
  /app Update.bat /hide
VERBOSE: StdOut and StdError redirection disabled
```

texhex, 2017-06-12:
Could you please try to install the "Intel Management Engine Components Driver" from https://support.hp.com/za-en/drivers/selfservice/hp-zbook-14-mobile-workstation/5403903 and then try to rerun BIOS Sledgehammer?

I really hope this is not the same as with the 6570b where it seems HP did not test if the firmware file can actually be installed.

bandwith, 2017-06-13:
Yes, after installing the "Intel Management Engine Components Driver" from http://ftp.hp.com/pub/softpaq/sp79001-79500/sp79059.html BIOS Sledgehammer is able to install the firmware update. 

Is there a method to chain the installation as a prerequisite using BIOS Sledgehammer?

texhex, 2017-06-13:
Good to hear, so at least it works for that model. 

What do you mean with "chain the installation"? You can of course install the driver before, this shouldn't be a problem. 

The silent install command should be in the CVA file for the SoftPaq. If you copy the download URL for the SPxxxx.EXE in question, just replace ``.EXE`` at the end with ``.CVA`` and you can view the file. In it, there is a setting "SilentInstall" which should give the command.

bandwith, 2017-06-13:
By "chain the installation", how could we use BIOS Sledgehammer to manage the installation of that specific SoftPaq and not have the need to install manually in advance? 

I haven't read the full script to determine if placing another .txt file with the appropriate download paths would work...

... more after I get back to the office next week.

texhex, 2017-06-19:
Thanks, understood. 

Supporting another TXT file that launches an EXE isn't that hard, this could be done in less than an hour. 

BUT:
Just creating a "ME-DriverInstall.txt" isn't something I want to do. The approach is to specialized for the ME updates IMHO. I could think of something like "DriverInstall-Pre1.txt" (install something before any firmware updates start) and "DriverInstall-Post1.txt" (after all updates were done). 

The problem with any driver install is: How can we detect that the driver needs installation of if it's already installed? Check file version? Registry? Both of them?

texhex, 2017-07-09:
@bandwith: Any updates?

bandwith, 2017-11-20:
Revisiting this issue today. More information soon.

texhex, 2018-01-19:
@bandwith I think I had an idea yesterday to solve this: The newer Intel SA * tools return the information called "Driver_Installed". 

We could use that information to decide if a driver installation is necessary or not, so the detection part would be easy. 

Is it possible that you could check with a ZBook 14 that, if the HP ME driver is not installed, the Intel SA tool repots "Driver_Install" = False?

texhex, 2018-03-27:
@bandwith I want to release 4.0 BETA next week and it would be great if you give me an update until then. Thanks.

texhex, 2018-04-06:
No updates, assuming this issue is abandoned and closing it.

closed 2018-04-06 by texhex