texhex/BiosSledgehammer #17: 3.1.2 quote

krevard, 2017-07-05:
Hi,
I had an error stating that BCU cannot communicate with the BIOS when getting UUID.
The "error" was on line 422. I deleted the single quote and left only double quotes.
$output=&BCU_EXE /GetValue:"$Name" | Out-String
With this correction it works now fine.
Let me know if u want more infos and thanks for your Work

texhex, 2017-07-05:
Thanks for the effor @krevard, but actually that are back ticks to prevent PowerShell from trying to escape the parameter (see [this question on SO](https://stackoverflow.com/questions/18116186/escaping-quotes-and-double-quotes)). 

They are there because some BIOS settings are not correctly escaped without them. I can't remember which ones that were, but without the back ticks it didn't worked. The problem with the script not recognizing the device should be fixed with the next release, it will also then test SERIAL NUMBER (see #16). 

The script directly in this repo already has this fix, but for a full release it might take until this weekend.

krevard, 2017-07-06:
Hi,
Thanks for your answer. For now I tested only an HP 840 G3.
I tested the command manually and the quote were needed (wihout them I have the same error manually or by your script)
BiosConfigUtility64.exe /getValue:"Universally Unique Identifier (UUID)"
I will test on other models as soon as the script work for this one.
My goal is to put your script in SCCM 2012.
Thanks.

texhex, 2017-07-09:
@krevard: I just released [v3.2.4](https://github.com/texhex/BiosSledgehammer/releases/tag/v3.2.4). In this version the script will try UUID first and if this comes up empty, I use the value SERIAL NUMBER for testing. Works on the two or three devices that show the UUID defect so I hope it will also work for you (fingers crossed ;-). 

Please let me know if this works for you.

krevard, 2017-07-10:
Hi,
It works a lot better now :-) thx
I have others "issues" as I m testing deeper. But I think I will open  other issues if u want.
The problem is th HP TPM updater cannot run in Winpe as stated in https://support.hp.com/lv-en/document/c05381064. When I use it true your script the following error is raised:
HPTPMFWUpdate Return Code = 262

The other thing is that I have a winpe 64 bit image and the Intel Me firmware update is a 32 application. Winpe does not have the wow64....
So I will have some Work to make it work with my SCCM task sequence.
Thanks for your help

texhex, 2017-07-10:
Good to hear, and regarding the TPM update and WinPE: 

>  The utility can only be run in Windows 7, Windows 8.x, or Windows 10. **It does not support Windows PE.** 

So the error message is "OK", since the tool is run on an unsupported platform. Or I have totally misunderstood you that you wanted it to run in WinPE.

krevard, 2017-07-10:
Yes it s ok to have an error message since it s not supported by HP in Winpe.
I will have to run it after the OS is installed. So I think I will have to split your script to run some parts in Winpe and some after the OS is installed.
The point is that your script "cannot work" as is in sccm or MDT in a staging phase. This due to HP application limitations. I wanted to let you know that.

texhex, 2017-07-11:
Thanks, now I think I understood what the problem was. I will add the following sentence to the System Requirements:
> Due to restrictions in several HP tools, it can NOT be run in Windows Preinstallation Environment (WinPE)

Does this make sense to you?

Also, which parts of it would you still want to run in PE/staging phase and why? This is just out of my curiosity.

krevard, 2017-07-14:
Yes it makes sense. 
The intel console tool doesn't work in winpe 64 (but should work in Winpe 32)
And the tpm update is not supported by HP in winpe. But as we see it can work :-)
So in Winpe (64) I'll run the config and update BIOS.
After the apply OS (SCCM Task sequence) I will try to update the TPM.
For the Intel AMT Firmware i already have a mechanism through SCCM applications that I add to my task sequence.
Thanks for your support.

texhex, 2017-11-05:
@krevard There were no updates on this for a long time and it seems to be working for you. Can we close this issue?

texhex, 2017-12-16:
@krevard Closing this issue now, please feel to reopen it if you think this is incorrect.

closed 2017-12-16 by texhex