texhex/BiosSledgehammer #18: Update TPM in UEFI/WinPE

SuperDOS, 2017-07-10:
Having issue updating TPM 1.2 to 2.0 on HP EliteBook 840 G3 in SCCM OSD.
First I format the disk with UEFI, also add a HP_TOOLS partition.

After updating TPM and the computer restarts and I press F1 to continue the update, I get "Firmware Image not Found".

This works fine when running legacy BIOS and not UEFI.

texhex, 2017-07-11:
That the TPM update does not run in WinPE was just discovered in #17, so this is interesting that it works for you but fails later on. 

Could you please post the log of the BiosSledge script with -Verbose activated, maybe we can get something from the file.

SuperDOS, 2017-07-11:
Well I got it works now. Works fine in WinPE even that HP states otherwise. So the thing I had to do was don't create a hp_tools partition instead the issue was that since I boot uefi with secure boot. HP looks for the firmware on the EFI partition which doesn't have the files since they are on the HP_Tools partition. For some reason the tool couldn't create the folder EFI\HP on the EFI partition so I run "cmd /c mkdir S:\EFI\HP before the TMP update runs. After reboot the firmware is found. If you run in legacy mode you might need the HP_Tools partition but should work using the option -c on the tpmconfigx64.exe. Funny thing is that HP says the -c option is ignored when having a GPT formated disk but I noticed it still creates it.

Hope this makes sense, typed it out on my phone from the top of my head. :)

texhex, 2017-10-08:
@SuperDOS This issue was now several months open and it seems the way you use Bios Sledgehammer works for you. There were no additional comments for this issue, would you be OK with closing this issue?

closed 2017-10-08 by SuperDOS