texhex/BiosSledgehammer #19: Change boot order

bartjanssen, 2017-10-02:
Wow this is a great piece of engineering. Thanks for sharing! 
Does anyone knows how to set boot order? I can not find this in the examples. 

Kind regards

texhex, 2017-10-08:
@bartjanssen You're welcome.

First, you should have one device that has the boot order (and the boot mode [UEFI/Legacy] set as you want to have it. 

Then execute "GetAllBiosSettings.bat" from the ``BiosSledgehammer\BCU-4.0.21.1\ `` folder. The resulting text file will include something like this (assuming the boot mode is UEFI):

```
UEFI Boot Order
	HDD:SATA:1
	HDD:USB:1
	NETWORK IPV4:EMBEDDED:1
	NETWORK IPV6:EMBEDDED:1
```

Append each line with a ``,`` and add it to BIOS-Settings.txt. It should then look like this:

``UEFI Boot Order == HDD:SATA:1,HDD:USB:1,NETWORK IPV4:EMBEDDED:1,NETWORK IPV4:EMBEDDED:2 ``

That should do the trick. Would be great if you could give it a try and let me know if this works as expected. 

A final note: There are reports from HP users that the [boot order settings might not stick on a ProBook](https://h30434.www3.hp.com/t5/Business-Notebooks/Setting-UEFI-boot-order-with-BCU-doesn-t-stick-on-Probook/td-p/5629404). I have no idea if this error still happens or if it applies to the EliteBook series.

texhex, 2017-10-31:
@bartjanssen Did the example helped and were you able to change the boot order as desired?

texhex, 2017-12-16:
@bartjanssen I assume this issue is resolved, so I'm closing it. Please feel free to reopen it if you think this is incorrect.

closed 2017-12-16 by texhex