texhex/BiosSledgehammer #21: Updates for INTEL-SA-00075 version 1.0.3.215 and other fixes

GregoryMachin, 2017-10-18:
Hi, 
I downloaded the latest version of INTEL-SA-00075 version 1.0.3.215. There have been some changes since the version 1.0.1.39, the name of the .XML  file and the content in the XML, these both caused fatal errors for me.  I also found an issue with machines that do not have Bitlocker enabled. 

I have updated the code please see the changes in my fork of you project.   https://github.com/GregoryMachin/BiosSledgehammer/tree/ISA75-update-and-other-fixes 

Please let me know what your opinion is on these changes. 

Regards
Greg

texhex, 2017-10-18:
@GregoryMachin Thanks for that, I will pull the changes the next time I work on this project, but this might take some days. 

And you are absolutely right, the SA-00075 code requires changes with the newest version from Intel. I guess I will completely change it since the generation of the XML file is by default in the same folder as the EXE is. This can be a problem for MDT/SCCM configurations I guess because the folder will be filled with computer reports. 

Could you please explain the changes for the BitLocker detection. I'm a little bit confused why this change is required as I have tested the script several times on system that did not had BitLocker enabled so it should have worked.

GregoryMachin, 2017-10-31:
@Texhex, 
My apologies for the delayed response. 

I find that the code errors on our windows 7 x64 machines without BitLocker as the EncryptionMethod property is missing in the object $drivestatus  

Here are the properties as they are on my windows 7 machine:

[DBG]: PS C:\WINDOWS\system32>> $drivestatus | Get-Member
 
 
   TypeName: Microsoft.Management.Infrastructure.CimInstance#root/CIMV2/Security/MicrosoftVolumeEncryption/Win32_EncryptableVolume
 
Name                      MemberType Definition                                                                                                                                               
----                      ---------- ----------                                                                                                                                               
Clone                     Method     System.Object ICloneable.Clone()                                                                                                                         
Dispose                   Method     void Dispose(), void IDisposable.Dispose()                                                                                                               
Equals                    Method     bool Equals(System.Object obj)                                                                                                                           
GetCimSessionComputerName Method     string GetCimSessionComputerName()                                                                                                                       
GetCimSessionInstanceId   Method     guid GetCimSessionInstanceId()                                                                                                                           
GetHashCode               Method     int GetHashCode()                                                                                                                                        
GetObjectData             Method     void GetObjectData(System.Runtime.Serialization.SerializationInfo info, System.Runtime.Serialization.StreamingContext context), void ISerializable.Get...
GetType                   Method     type GetType()                                                                                                                                           
ToString                  Method     string ToString()                                                                                                                                        
DeviceID                  Property   string DeviceID {get;}                                                                                                                                   
DriveLetter               Property   string DriveLetter {get;}                                                                                                                                
PersistentVolumeID        Property   string PersistentVolumeID {get;}                                                                                                                         
ProtectionStatus          Property   uint32 ProtectionStatus {get;}                                                                                                                           
PSComputerName            Property   string PSComputerName {get;}        

As the property EncryptionMethod is missing this causes the script to error. 
Regards
Greg

texhex, 2017-11-01:
@GregoryMachin No problem.

Thanks for the clarification, it can really happen that my Windows 7 test there was incorrect. Please give me some days to check this, I wasn't able to work on it today since the installer for Hack required some attention. 

I will check this next weekend, I hope I can borrow a Windows 7 test machine until then.

texhex, 2017-11-02:
I had the chance to test three computers today and the results vary a lot:
* Laptop: Was as how I expected it, EncryptionMethod exists and returned 0
* Desktop 1: In for reinstall as "WMI is broken". Had EncryptionMethod  property but returned NULL as result
* Desktop 2: As you have described, script crashed as Encryption Method was missing. However, this one was in for an activation error so maybe this was no longer considered "Windows Enterprise" and had no access to BitLocker?

Based on that, I added the code from you and mixed some additional checks in. This code that should be able to prevent any error in the future, I did run it successfully on all three machines.

Let me know what you think.

```powershell
        #Because we do not know if we can access the BitLocker module, we try to figure it out using CIM/WMI
        $encryptableVolumes=Get-CIMInstance "Win32_EncryptableVolume" -Namespace "root/CIMV2/Security/MicrosoftVolumeEncryption"
                      
        $systemdrive=$env:SystemDrive
        $systemdrive=$systemdrive.ToUpper()

        Write-Verbose "Testing if system drive $($systemdrive) is BitLocker encrypted"

        if ( $encryptableVolumes -ne $null) 
        {
            foreach($drivestatus in $encryptableVolumes)
            {
                if ( $drivestatus.DriveLetter.ToUpper() -eq $systemdrive)
                {           
                     Write-Verbose "Found entry for system drive"
                               
                     #The property ProtectionStatus will also be 0 if BitLocker was just suspended, so we can't use this property.
                     #We go for EncryptionMethod which will report 0 if no BitLocker encryption is used

                     #As @GregoryMachin reported, some versions of Windows do not have this property at all so need to make sure to have access to it
                     #See https://github.com/texhex/BiosSledgehammer/issues/21
                     if (Get-Member -InputObject $drivestatus -Name "EncryptionMethod" -Membertype Properties)
                     {
                        #Some computers reported NUL/NIL as the output, so we need to make sure it's uint32 what is returned
                        #See https://msdn.microsoft.com/en-us/library/windows/desktop/aa376434(v=vs.85).aspx
                        if ( $drivestatus.EncryptionMethod -is [uint32])
                        {
                            if ( $drivestatus.EncryptionMethod -ne 0 )
                            {
                                $bitLockerActive=$true
                                write-host "BitLocker is active for system drive ($systemdrive)!"
                            }                        
                            else
                            {
                                Write-Verbose "BitLocker is not used"
                            }
                        }
                    }
                }
            }
        }
```

texhex, 2017-11-06:
@GregoryMachin Updates for new ISA-75DT are now also included, please have a look at the commits (above) and let me know what you think.

texhex, 2017-11-07:
[Version 3.3.0](https://github.com/texhex/BiosSledgehammer/releases) with these issues fixed is now available.

GregoryMachin, 2017-11-08:
@texhex Thank you for the fix, looks good. I'll update my local repo and test.

closed 2017-11-08 by GregoryMachin