texhex/BiosSledgehammer #28: Network boot...

ofurpesi, 2017-11-06:
Hi and thank you for a great tool :)

I´m hoping anyone here can help me. I´m using mdt to deploy Windows 10. After mdt deploys bios settings, it reboots and goes straight to network boot. I have no idea why. Any ideas?

texhex, 2017-11-06:
Please attach the BIOS-Settings.txt you are using and also the log output of BIOS Sledgehammer so I can get an idea where this problem comes from.

ofurpesi, 2017-11-07:
Here is the bios settings file
[BiosConfigProdesk600G1.txt](https://github.com/texhex/BiosSledgehammer/files/1449216/BiosConfigProdesk600G1.txt)

Can you please explain where i can find the log output of sledgehammer?

texhex, 2017-11-07:
OK, and the file is called how on your installation in which folder? Because that filename will never be used by BIOS Sledgehammer so the file would be ignored. 

How do you boot into MDT - using a USB stick or Network boot? Also, does the machine at least boot once from C: before executing BIOS Sledgehammer or do you try to change the BIOS Settings in the PE (first) phase? 

The location of the log file is described in readme.md (directly below the file list when you click on the "Code" tab in this repo).

texhex, 2017-12-11:
As there was no response in a long time, I'm closing this issue. Please feel free to reopen it again if you think this incorrect.

closed 2017-12-11 by texhex