texhex/BiosSledgehammer #29: Add parameters of the various resources

GregoryMachin, 2017-11-10:
@texhex  
Hi, I trust you are well. 

I was looking to make the script a bit more dynamic by having parameters for the various resources 
-BCUpath 
-IAS75Path
-PwdFilesPath
-ModulesPath
As I don't have an HP machine at home this weekend so I cant test the changes so they are staged on a branch for now. I'll try and get the testing done on Monday 
https://github.com/GregoryMachin/BiosSledgehammer/tree/Add-paramters 

Let me know if this is something you are willing to incorporate into the script, if so I'll also ammend the documentaion. 

Regards

texhex, 2017-11-12:
@GregoryMachin Thanks for that and of course you can do whatever you want in your downstream repo, but I do not think I would accept a PR with that changes for the following reasons:
* Making those paths variables that could be passed on would mean that we expect the results of at least BCU and IS75DT to be deterministic. Regardless of which version the operators gives us, the script should work as before. As you know yourself with the updates of ISA75DT, this is not true. I think I made a total of two script changes because BCU was changed by HP and did no longer work. Hence, given that a new version of those tools also require a script change, they should remain inside the script directly so if someone wants to use a new version, they should be able to edit the source script. 
* Also, if those paths should be configurable by the operators, we should stick to the existing method that we use for the firmware configuration: TXT files with Key==Value entries This is to keep the methods consistent. 
* I do not think that most operators care about those paths much. I might be wrong, but I guess most people that download it (aren't that much after all ;-) just want to have a solution for all that firmware updates scripting, they don't actually care if we are using the newest BCU or ISA75DT tool.

GregoryMachin, 2017-11-12:
@texhex thanks to taking time to consider my PR and providing a great response. 

I  can see how how it could cause a large overhead for managing the code.

I expect my needs are a little different to most users, as I have a large fleet of machines with a large number of models, thus looking to decouple the script from the rest so I can manage them independently. 

I'll look for a way to manage it inhouse. 

Thank you.

texhex, 2017-11-13:
You're welcome. Could you maybe explain why it is not enough to have one base folder for all the models? At work, we have data for about 25 models in a single \Models folder and so far I did not come across any limitation. Or do you require separate settings for the same model?

GregoryMachin, 2017-11-15:
There are a number things I was considering:
- I'm a little concerned about the size of the package if I were to included the bios and other firmware for all the modes. 
- I was looking to manage the configurations separately to the script code so that there is a clear division. 
- having a repository for each series eg Elitebook 800 G2 series  that has the firmware (bios, ME, TPM), and bios configurations. 
- the above allows me to reduce the package size, reduce the risk when implementing changes, be more targeted with implementing changes. We have stringent ITIL change control process at work.   

I would be interested to hear how you manage your environment.

texhex, 2017-11-20:
Well, we have one installation where all files go into, so to speak a centralized \Models folder. Size is currently about 800MB. 

If you keep the changes in git, I do not see any reason why not use one repo instead of several others. At the end, you can always push per "model" with a reference to the documented change (which is documented in another tool I guess). 

The only advantage I can see right now is a faster incident response: Failure for model EB 800 G2, locate repo "EB800G2", see push two days ago, must be because of CHxxxx. 

But on the other hand, I would expect the external tool to make that connection. If the configuration item model is of EB800G2 and there was a change of EB 800 G2, I actually do not need to check the repo as I see the recent changes directly (from the change board). 

Or I totally misunderstood what you meant with that, I'm a newbie to ITIL.

texhex, 2017-12-16:
@GregoryMachin Were my comments to totally off that you gave up on this issue? ;-)

texhex, 2018-01-18:
@GregoryMachin I would like to close this issue in the next days. Is this OK with you?

GregoryMachin, 2018-01-20:
@Micheal Hex
Yes, please close.

Currently I have a package staged for the Meltdown and Spectre Bios
patching, I only have 1/3 to 1/2 of the models in the package and the
package is over 700MB note this is only the Bios Firmware. Given I have to
target over 5000+ machines its going take a while to get it rolled out. I'm
going to split up the package once I know the full size otherwise it will
take to long to distribute and probably load the network.

Once we have this roll out sorted I will look at moving to using this as
our primary method of deploying bios configurations and firmware.

Regards

Greg

texhex, 2018-01-20:
@GregoryMachin Thanks, will do so. 

Just for your information, please note that Intel has confirmed [reports that the current microcode updates cause unexpected reboots](http://www.zdnet.com/article/meltdown-spectre-intel-says-newer-chips-also-hit-by-unwanted-reboots-after-patch/) for ALL CPUs (not just Broadwell and Haswell as reported earlier). Given that G1/G2 devices also have a BIOS downgrade lockout policy, once you have rolled out those updates and want to roll back because of problems, you won't be able to do so.

Because of this, our company has decided to NOT roll out any microcode updates until the quality has reach a better level.

Just for your information.

Closing issue.

closed 2018-01-20 by texhex

texhex, 2018-01-20:
@GregoryMachin And just as a sidenote: As HP plans to produce more form factors based on the same platform (as they do with the EliteBook 8x0 models or ZBook 1x today), the size of the update files will become an issue. I plan to address this with the v4.x release to be able to use shared file locations. In case you have some ideas how to achieve this and want to share, please feel free to open a new issue.