texhex/BiosSledgehammer #30: TPM upgrade to 2.0 instead of 1.2

PhilippRC, 2017-12-11:
Desired TPM Spec is 1.2 but the chosen upgrade file is to version 2.0:

***** TPM Update ************************************************
Reading TPM update information from [C:\Temp\BiosSledgehammer\Models\HP EliteBook 840 G3\TPM-Update.txt]...
VERBOSE: Reading hashtable from C:\Temp\BiosSledgehammer\Models\HP EliteBook 840 G3\TPM-Update.txt
TPM manufacturer ID of this device: 1229346816
TPM manufacturer ID required......: 1229346816
  TPM manufacturer IDs match
Current TPM Spec: 1.2  (Numerical: 1.2)
**Desired TPM Spec: 1.2  (Numerical: 1.2)**
  TPM Spec version matches
Current TPM firmware: 6.40  (Numerical: 6.40)
Desired TPM firmware: 7.62  (Numerical: 7.62)
  Firmware version is lower than desired, update required
Update required result:
  Update required because of TPM Spec....: False
  Update required because of TPM firmware: True
TPM update required!
Searching firmware file entry for [6.40]...
Firmware file found:
  [Firmware\TPM12_6.40.190.0_to**_TPM20**_7.62.3126.0.BIN]

texhex, 2017-12-11:
Yes, because your TPM-Update.txt is incorrect.

Each firmware entry in TPM-Update.txt define which file to hand over to the updater exe depending on which firmware the device is currently on. In your example, TPM-Update.txt defines that a device with 6.40 should use the file ``TPM12_6.40.190.0_to_TPM20_7.62.3126.0.BIN``. 

See the docs here: https://github.com/texhex/BiosSledgehammer#tpm-update

I guess you used one of the example files and just changed the spec version. All our example files are designed for TPM 1.2 to 2.0 updates; if you want to stay on 1.2 you need to define the firmware files for 1.2. The HP advisory lists all of them: https://support.hp.com/us-en/document/c05381064/

PhilippRC, 2017-12-12:
Thanks, I understand now. Since the two spec versions have different firmware versions, to define a target firmware version of 7.62 already implies that spec version has to be 2.0.

closed 2017-12-12 by PhilippRC