texhex/BiosSledgehammer #31: Executing BCU command with ` fails on Windows 8.1

PhilippRC, 2017-12-11:
I know these quotes were already mentioned before (Issue: "3.1.2 quote") but the comment sais it is resolved.  However, for me it is not working:

Line 463 and 464 in BiosSledgehammer.ps1:
#$output=&$BCU_EXE -getvalue $Name | Out-String
$output=&$BCU_EXE /GetValue:`"$Name`" | Out-String

BCU command works when inversing the comments of the two lines.

Verifying BIOS Configuration Utility (BCU) can communicate with BIOS.
  Trying to read Universally Unique Identifier (UUID)...
VERBOSE: Reading BIOS setting using different setting names
VERBOSE:    Trying using setting name [Universally Unique Identifier (UUID)]...
VERBOSE: Read BIOS value: Result from BCU =============
VERBOSE: <BIOSCONFIG Version="" Computername="PCS10778" Date="2017/12/11" Time="14:46:06" UTC="1">
	<SETTING changeStatus="warning" name = "Universally" returnCode="20">
		<VALUE><![CDATA[]]></VALUE>
	</SETTING>
	<Information msg="BCU return value" real="0" translated="0" />
</BIOSCONFIG>
VERBOSE: ==============================================
VERBOSE: Value:
VERBOSE: Return code: 20
VERBOSE:    Trying using setting name [Enter UUID]...
VERBOSE: Read BIOS value: Result from BCU =============
VERBOSE: <BIOSCONFIG Version="" Computername="PCS10778" Date="2017/12/11" Time="14:46:08" UTC="1">
	<SETTING changeStatus="warning" name = "Enter" returnCode="20">
		<VALUE><![CDATA[]]></VALUE>
	</SETTING>
	<Information msg="BCU return value" real="0" translated="0" />
</BIOSCONFIG>
VERBOSE: ==============================================
VERBOSE: Value:
VERBOSE: Return code: 20
VERBOSE:    Trying using setting name [Universal Unique Identifier(UUID)]...
VERBOSE: Read BIOS value: Result from BCU =============
VERBOSE: <BIOSCONFIG Version="" Computername="PCS10778" Date="2017/12/11" Time="14:46:09" UTC="1">
	<SETTING changeStatus="warning" name = "Universal" returnCode="20">
		<VALUE><![CDATA[]]></VALUE>
	</SETTING>
	<Information msg="BCU return value" real="0" translated="0" />
</BIOSCONFIG>
VERBOSE: ==============================================
VERBOSE: Value:
VERBOSE: Return code: 20
  Failed.
  Trying to read Serial Number (S/N)...
VERBOSE: Reading BIOS setting using different setting names
VERBOSE:    Trying using setting name [Serial Number]...
VERBOSE: Read BIOS value: Result from BCU =============
VERBOSE: <BIOSCONFIG Version="" Computername="PCS10778" Date="2017/12/11" Time="14:46:11" UTC="1">
	<SETTING changeStatus="warning" name = "Serial" returnCode="20">
		<VALUE><![CDATA[]]></VALUE>
	</SETTING>
	<Information msg="BCU return value" real="0" translated="0" />
</BIOSCONFIG>
VERBOSE: ==============================================
VERBOSE: Value:
VERBOSE: Return code: 20
VERBOSE:    Trying using setting name [S/N]...
VERBOSE: Read BIOS value: Result from BCU =============
VERBOSE: <BIOSCONFIG Version="" Computername="PCS10778" Date="2017/12/11" Time="14:46:12" UTC="1">
	<SETTING changeStatus="warning" name = "S/N" returnCode="20">
		<VALUE><![CDATA[]]></VALUE>
	</SETTING>
	<Information msg="BCU return value" real="0" translated="0" />
</BIOSCONFIG>
VERBOSE: ==============================================
VERBOSE: Value:
VERBOSE: Return code: 20
  Failed.
\\gva\DfsData\GVA\Data\Common\24h\euc\BiosSledgehammer\BiosSledgehammer.ps1 :
Unable to communicate with BIOS. Can't continue.
    + CategoryInfo          : NotSpecified: (:) [Write-Error], WriteErrorExcep
   tion
    + FullyQualifiedErrorId : Microsoft.PowerShell.Commands.WriteErrorExceptio
   n,BiosSledgehammer.ps1

\\gva\DfsData\GVA\Data\Common\24h\euc\BiosSledgehammer\BiosSledgehammer.ps1 :
Unable to start (see previous errors)
    + CategoryInfo          : NotSpecified: (:) [Write-Error], WriteErrorExcep
   tion
    + FullyQualifiedErrorId : Microsoft.PowerShell.Commands.WriteErrorExceptio
   n,BiosSledgehammer.ps1

BIOS Sledgehammer finished, return code 666.

texhex, 2017-12-11:
Hmm, this is strange. Could you please start a new PowerShell prompt and enter ``$psversiontable`` and post the results here?

PhilippRC, 2017-12-12:
Here is the output of $psversiontable:

Name                           Value
----                           -----
PSVersion                      4.0
WSManStackVersion              3.0
SerializationVersion           1.1.0.1
CLRVersion                     4.0.30319.36366
BuildVersion                   6.3.9600.18773
PSCompatibleVersions           {1.0, 2.0, 3.0, 4.0}
PSRemotingProtocolVersion      2.2

texhex, 2017-12-16:
I tried to reproduce the problem here, but I'm unable to find out what exactly Windows 8 does not like about this command (I only have access to Windows 7 and Windows 10). The command works on Windows 7 plus PS 4.0, which is exactly the same version you have. 

I also rechecked the [docs](https://docs.microsoft.com/en-us/powershell/module/microsoft.powershell.core/about/about_quoting_rules?view=powershell-4.0) and don't see any issue there. 

If we use the the "old" command (without backticks), this will cause BCU to fail because in this case PS will not create one parameter ``/GetValue:Universally Unique Identifier (UUID)`` but instead split the parameters on each space. At least that's what has happened in the past. 

All I could offer right now is to remove the sentence that it should work on Windows 8 completely from the docs.

texhex, 2018-01-18:
@PhilippRC 
I have tried to get my hands on a Windows 8 machine, but failed to do so. 

I also thought about adding some sort of OS depending call but as I have used the ` escaping often, and given that I believe Windows 8 is not that often used, I don't think this is worth the effort. 

Therefore I would like to close this issue. Is this OK from your side?

PhilippRC, 2018-01-19:
Thanks for the effort.
Our focus is already shifting to Win 10 which will be deployed in a few months, so yes closing this Win 8.1 issue is fine with me. I will try BiosSledgehammer with Win10 later on.

closed 2018-01-19 by PhilippRC