texhex/BiosSledgehammer #33: Why decrypting Bitlocker if Microsoft recommends only to suspend?

PhilippRC, 2017-12-11:
Isn't it enough just to suspend bitlocker for the TPM firmware update?
According to MS it is okay: https://technet.microsoft.com/en-us/library/hh831507(v=ws.11).aspx

To fix the Infineon TPM vulnerability (ADV170012.), also a Bitlocker suspend and Clear TPM is needed after the TPM firmware update (for Bitlocker only TPM 1.2 is affected):
https://support.microsoft.com/en-us/help/4046783/bitlocker-mitigation-plan-for-vulnerability-in-tpm

texhex, 2017-12-11:
First because that way it's 99.9% secure secure that we do not FUBAR the machine (firmware issues could still trash the machine), because somehow the TPM protector is still attached and the user has no recovery key. 

Second, because some versions of the TPM update utilities check if BitLocker is off (not just suspended) and will not allow an update. 

However, there is the option that the BitLocker is not decrypted, please see the docs: https://github.com/texhex/BiosSledgehammer#disable-automatic-bitlocker-decryption

PhilippRC, 2017-12-12:
Thanks for the feedback.

closed 2017-12-12 by PhilippRC