texhex/BiosSledgehammer #34: Mass of BIOS Upgrades incoming.

bandwith, 2018-01-09:
https://support.hp.com/us-en/document/c05869091?hprpt_id=HPGL_ALERTS_1996200&jumpid=em_alerts_us-us_Jan18_xbu_all_all_1415449_1996200_LaptopsandHybridsDesktopsWorkstationsPointofSaleSystemsTablets_critical__/

texhex, 2018-01-10:
Thanks. Checking the CVE number noted there, it seems this to be updates for Spectre. 

Happy to merge a PR from you in case you plan to update your devices.

bandwith, 2018-01-10:
Working on our models. A number of the published SoftPaqs aren't available
for download yet.

bandwith, 2018-01-11:
Found that the notice had a few typos in the download URLs published. Once converted to lowercase, and a few verified against the listing at support.hp.com for the individual models, the ``SPaqURL`` are working.

texhex, 2018-01-14:
I had some time yesterday morning and pulled the data for most of the included models. However, it would be great if you could create a PR for 2570p, 6570b and 8570p - they should all be available at the end of this month.

Do you have anywhere a Folio 1040 G2 around? The download HP offers seems to be incorrect, as this is for the M77 BIOS type while the last download for the G2 we had was M76 so I think this is incorrect.

This is the list of devices that I was able to include:
* EliteBook 820 G1, EliteBook 840 G1, EliteBook 850 G1
* EliteBook 820 G2, EliteBook 840 G2, EliteBook 850 G2
* EliteBook 820 G3, EliteBook 840 G3, EliteBook 850 G3
* EliteBook 820 G4, EliteBook 840 G4, EliteBook 850 G4
* EliteBook Folio 1040 G1, EliteBook Folio 1040 G3
* EliteBook x360 1030 G2
* Pro x2 612 G2
* ProDesk 600 G1 TWR
* ProDesk 600 G2 SFF
* ProDesk 600 G3 MT, ProDesk 600 G3 SFF
* Z440 Workstation
* ZBook 15 G2
* ZBook 15 G3, ZBook 17 G3

bandwith, 2018-01-14:
We don't. We're a relatively new HP shop. 

I'm still working on some validation before the PR is submitted, but we have the following models:

- EliteBook 1040 G4
- Z1 G2 Touch Workstation
- Z230 SFF Workstation
- Z240 SFF Workstation
- ZBook 14 (G1)
- ZBook Studio G3

texhex, 2018-01-14:
OK, thanks. 

Those models would be a great addition, looking forward for the PR!

texhex, 2018-03-27:
@bandwith Given that the Intel fix required fixing itself, I will go through the entire BIOS list on the next weekend again and include the BIOS versions again. Do you think you have your list ready until then?

texhex, 2018-04-06:
No updates, assuming this issue is abandoned and closing it.

closed 2018-04-06 by texhex