texhex/BiosSledgehammer #37: Support Intel SA-00086 detection for ME firmware

texhex, 2018-01-19:
In 2017-11 another ME security bug was found: Intel SA 00086 / HPSBHF03571 (aka CVE-2017-5705, CVE-2017-5706, CVE-2017-5707, CVE-2017-5708, CVE-2017-5709, CVE-2017-5710, CVE-2017-5711)

* [HP Inc. Advisory HPSBHF03571](https://support.hp.com/us-en/document/c05843704)
* [Intel Security Advisory](https://security-center.intel.com/advisory.aspx?intelid=INTEL-SA-00086&languageid=en-fr)
* [Intel Support Document](https://www.intel.com/content/www/us/en/support/articles/000025619/software.html)
* [Intel Detection Tool](https://downloadcenter.intel.com/download/27150?v=t)

Given that this bug is newer then the one noted in [HPSBHF03557/ Intel SA 00075](http://www8.hp.com/us/en/intelmanageabilityissue.html), we believe a system that is secured against SA 00086 is also secured against SA 00075. 

We should therefore replace the SA-75 detection tool with the SA-86 detection tool and update the ME firmwares with new firmware files.

GregoryMachin, 2018-01-24:
@texhex 
Hi, 
I posted a query on the Intel forums and the result is that each of the SA tools is a stand alone tool.  

https://communities.intel.com/message/521691#521691

Regards
Greg

texhex, 2018-01-24:
@GregoryMachin Dang it. This would mean that running the SA-86 tool on a system that is affected by SA-75 but not SA-86 would result in the tool reporting "All good". 

I think I will use the 4.0 rewrite to strip down the ME part that we ONLY use the ME version to detect if an update is necessary or not, as we already do with BIOS and TPM updates (No special "Is Vulnerable" detection).

Thanks for pointing this out! 

P.S.: Awesome username you use on the Intel site : )

GregoryMachin, 2018-01-24:
Thanks, yeah been using the username since early 2000s and have seen anyone
else using it lol.

It's annoying it would have been good if they where accumulative. Would
need to run them both for true results.

 I agree used the same methode as the bios and tpm.

G

texhex, 2018-03-27:
@GregoryMachin 4.0 BETA code is now online (no release so far, as the documentation is missing) but the changes are included. 

I will check if I also include the changes proposed #36 so a on-the-fly BIOS change for ME will also be added. 

Closing this issue now, please reopen if you think this requires more attention.

closed 2018-03-27 by texhex