texhex/BiosSledgehammer #39: Support new "Physical Presence Interface" BIOS setting

texhex, 2018-02-20:
According to a discussion on the [HP forums](https://h30434.www3.hp.com/t5/Business-Notebooks/Upgrading-TPM-Firmware-version-silently/td-p/6173587) there is now a setting in newer BIOS versions that can disable the extra "Press F1 to confirm" messages during TPM update.

The setting seems to be
```
Physical Presence Interface
```
With Disable or Enable setting.

This could be added to ```TPM-BIOS-Settings.txt```

texhex, 2018-03-13:
See also: https://community.spiceworks.com/topic/1836570-hp-tpm-configuration-utility-updating-tpm-firmware-and-converting-between-tpm?page=2

texhex, 2018-04-11:
Also https://h30434.www3.hp.com/t5/Business-Notebooks/Physical-Presence-Interface/td-p/6474110

datagutten, 2018-04-20:
I tried it on a ProBook 430 G3 where it required me to enter a four digit code to disable presence.
The prompt appeared after the prompt for TPM update, so placing it in the TPM settings file seems pretty useless.
To have any effect the setting should be applied and the computer rebooted before the TPM update is applied.

texhex, 2018-04-23:
I had a longer discussion about this setting with a HP technician last week, especially because there is zero documentation available. He explained it that the PPI expects this code because the overall security of the BIOS itself is lowered as all actions would he "hidden" - so this would explain that a code is needed.

You are absolutely right, it does not make sense to include this setting unless there are two or three changes outstanding. This is not possible at this time, as each section is processed independently by the script. But on the other hand: I do not now know if it makes more sense to have to enter a code once, or just press F1 two or three times. The "Fully silent" install it broken in any case.

However, he hinted at something: According to him "TPM Activation Policy" set to NO PROMPTS would prevent the TPM update confirmation. Do you have any test machines that require a TPM update where you could test that? Maybe we could get rid of the TPM update message that way.

datagutten, 2018-04-24:
I tested on a ProBook 640 G2 where I added `TPM Activation Policy==No prompts` to `TPM-BIOS-Settings.txt`. Then TPM was upgraded without questions.

texhex, 2018-04-25:
Thanks for the test and good to hear it works. I will check in the next days for all models that support the SLB9670 update to include this setting in TPM-BIOS-Settings.txt. 

I will have access to some more test devices next week and once the change is included, will release 4.1.

texhex, 2018-05-07:
[Version 4.0.6](https://github.com/texhex/BiosSledgehammer/releases/tag/v4.0.6) was released yesterday, closing this issue.

closed 2018-05-07 by texhex