texhex/BiosSledgehammer #40: Start-TranscriptIfSupporterError

Joseahan, 2018-03-06:
Hello!

Trying to run the script in a SCCM TS but i get this error when it runs:

Start-TranscriptIfSupported: A parameter cannot be found that matches parameter name 'File'.
\MPSXM.psm1: 552 char:5
Start-TranscpritIfSupported -Path $logPath -Name $logName -NewLog
CategoryInfo : AnvalidArgument: (:) [Start-TranscpirtIfSupported]. ParameterBindigException
FullyQualifiedErrorId : NamedParameterNotFound,Start-TranscpirtIfSupported

I've checked the code but cant find why i get this error, have any idea of what it could be?

texhex, 2018-03-06:
Well, that's interesting, thanks for the report. 

There is nothing you could do, this seems like an internal error within MPSXM.psm1 (the helper module included). 

BIOS Sledgehammer uses the function Start-TranscriptTaskSequence() which uses the function Start-TransscriptIfSupported() and somewhere there the error happens. I can not make any sense from the error message, as the function do not call themselves with a -File parameter.

The only thing I could imagine would be that the COM Object we are using from SCCM to get the log path is broken so it returns null for the path which causes this error. But this is not very likely. 

Can you maybe execute BIOS Sledgehammer using the batch file "RunVisible-Verbose.bat" and post a screenshot of the output (a smartphone capture will do as well)? This command displays more output which could be helpful to detect what is going on.

Joseahan, 2018-03-07:
This is what it looks like when i run "RunVisible-Verbose.bat":
[image]

So it's basically the same as i wrote in the first report.

texhex, 2018-03-07:
Thanks for that. The screenshot helps that the error is not related to the SCCM log path being empty that I first suspected. 

You have renamed the script name, this was never tested but I can't think this to be the error. I added additional verbose output so also the file name of the log will be written.

Please do the following:
* Download the entire repo using this link: https://github.com/texhex/BiosSledgehammer/archive/master.zip
* Extract the file MPSXM.psm1 file and copy it to the location where BIOS Sledgehammer is located, overwriting the existing file. 
* Then execute RunVisble-Verbose.bat again and post a screenshot.

Joseahan, 2018-03-08:
I downloaded the .zip that you linked, and replaced all the files so it was "fresh", and it looks like this:
[image]

Joseahan, 2018-03-08:
I found the issue, as it says it tries to create the log file in X:\Windows\temo\smstslog, although i have placed the step later in the TS when it runs in the current OS. If i place the step when it is still in PE the script runs but fails at "Test-Environment".
From what i remember you can't run the upgrade tool in WinPE?

texhex, 2018-03-08:
Thanks, but to be honest I still have no idea where this "-File" parameter error comes from. It should clearly run without issues within the OS itself. That it fails there but works in PE is really strange. 

And yes, some HP fail or do not work correctly in PE. Some of the them work fine in PE, but I never tested that and can't help you there. 

I will do some testing tomorrow, maybe the functions freaks out when a folder is read only or does not exist.

texhex, 2018-03-09:
I was able to identify the issue here. The function Start-TranscriptIfSupported() has one line to find out which log files already exist so it can create the next one. This line looks like this:

```
$existing_files = Get-ChildItem -Path $Path -File -Filter $filter -Force 
```

There we have the -File parameter but if the folder ``$Path`` exists, there will be no error. However, if the folder can not be found this PowerShell cmdlet will generate exactly the error message you are seeing: *A parameter can not be found that maches parameter FILE*.

I have now added two fail safes:
* The script checks if the given folder (e.g. X:\WINDOWS\TEMP\SMSTsLog) exists. If not, it will write an error and default to the TEMP folder (*Logfile path [X:\WINDOWS\TEMP\SMSTSLog] does not exist, defaulting to [C:\Users\USER~1\AppData\Local\Temp]*)
* If the above mentioned Get-ChildItem generates an error, the script will add this error message with more details so it's clear that a file listing has failed (*Unable to list files in path [X:\WINDOWS\TEMP\SMSTSLog] with filter [MPSXM-Debug_Tests.ps1.log-??.txt]: A parameter cannot be found that matches parameter name 'File'.*)

The change is now live, please do the same as before:

* Download the entire repo using this link: https://github.com/texhex/BiosSledgehammer/archive/master.zip
* Extract the file MPSXM.psm1 file and copy it to the location where BIOS Sledgehammer is located, overwriting the existing file.
* Then execute RunVisble-Verbose.bat again 

If everything works as I hope, you might still get an error (because the log path does not exist), but BIOS Sledgehammer should finally run.

Joseahan, 2018-03-13:
Now it works, thanks!

texhex, 2018-03-13:
Good to hear, thanks for your patience. Closing issue.

closed 2018-03-13 by texhex