texhex/BiosSledgehammer #41: TLS/HTTPS issue?

napzero, 2018-03-09:
I tried running StartExampleDownloads.bat and got this error...

> Start-DownloadFile : Exception calling "DownloadFile" with "2" argument(s): "The underlying connection was closed: An unexpected error occurred on a send." At C:\temp\BiosSledgehammer_v3.3.2\StartExampleDownloads.ps1:175 char:19
> + ... $tempFile=Start-DownloadFile -URL $URL -DownloadPath $DownloadPath
> +               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> + CategoryInfo          : NotSpecified: (:) [Start-DownloadFile], MethodInvocationException
> + FullyQualifiedErrorId : WebException,Start-DownloadFile

- I tried on another PC and on a VM and got the same.
- I got one download to work by changing the download URL from HTTPS to HTTP.
- I got HTTPS to work by adding the below code right before $webClient.DownloadFile($URL, $tempFile)

`System.Net.ServicePointManager]::SecurityProtocol = [System.Net.SecurityProtocolType]'Tls,Tls11,Tls12'`

Thoughts? 
BTW I'm new to GitHub, sorry if I'm not not using good form or something.

texhex, 2018-03-09:
Regarding if you did everything right: You included all details, confirmed that it was not single machine error and even included a working solution. If all issue reports would be like this, my life would be a lot easier. So, yes you did everything right 👍 

I did some in testing regarding this and the base issue is that PowerShell (by default) uses SecurityProtocolType.Tls|SecurityProtocolType.Ssl3 (as this the default in .NET 4.0/4.5) and ftp.hp.com has disabled both protocols as they are unsecure. See http://ssl-checker.online-domain-tools.com/ and enter ftp.hp.com to check which protocols and cipher suites they offer

Your code did the right thing because it turned on TLS 1.1 and 1.2. My code is just a little deviation of yours as it turns on TLS 1.1 and 1.2 without touching the default protocols. If TLS 1.3 ever becomes standard and is supported by .NET, that code should not overwrite it. 

```
[System.Net.ServicePointManager]::SecurityProtocol = 

[System.Net.ServicePointManager]::SecurityProtocol -bor
[System.Net.SecurityProtocolType]'Tls11' -bor 
[System.Net.SecurityProtocolType]'Tls12'
```

A fix was just pushed. Please do the following:
* Download the current state of this repo using this link: https://github.com/texhex/BiosSledgehammer/archive/master.zip
* Copy the file ``StartExampleDownload.ps1`` from the archive
* Copy it your current installation folder, overwriting the existing file
* Then please try again. 

NOTE: You will get download errors for several BIOS downloads for sure as it seems that HP has removed all BIOS versions without Spectre Fix. This will be the next task on my TODO list.

napzero, 2018-03-09:
Awesome! Thank you, and likewise I'm glad to have helped.

closed 2018-03-09 by napzero