texhex/BiosSledgehammer #42: Execution blocked from user temp

napzero, 2018-03-09:
This issue isn't a fault of BiosSledgehammer, but on our corporate network we have a GPO that prevents execution of programs from a user's local temp folder. This causes an error for us, which of course stops the entire thing from working: 

> 'BiosConfigUtility64.exe' failed to run: This program is blocked by group policy.

I tried a quick fix by swapping out entries of `Get-TempFolder` with `"c:\temp"` and it worked.

Would you do something like with the download script (c:\somefolder) or would it be reasonable to have a configurable temp folder location?

texhex, 2018-03-10:
As a side note: When using AppLocker, you should define at least one account where the restrictions do not apply at all. Else, if your GPO Administrator ever gets it wrong, you'll get in real trouble. 

I generally dislike the idea of passing configuration parameters to BIOS Sledgehammer except it is impossible that the script can determine the setting automatically. 

However, I have changed all direct calls to Get-TempFolder() with a constant that is defined on line 90 (see https://github.com/texhex/BiosSledgehammer/blob/89090eedf6a267652b484c272101cb40ffe5e12b/BiosSledgehammer.ps1#L90). You can simply remove the comment for line 92 and all calls that use TEMP_FOLDER will then use that folder instead of the TEMP folder from the user profile. 

The change is live in [release 3.4.0](https://github.com/texhex/BiosSledgehammer/releases). Please note that 3.4.0 uses a newer BCU and MPSXM versions, so that files/folders need to be copied as well.

napzero, 2018-03-11:
I don't think we are using AppLocker, but the note is quite right. 

Thank You!

closed 2018-03-11 by napzero