texhex/BiosSledgehammer #43: Bitlocker Decryption Error

Joseahan, 2018-03-13:
Hello!

Have another error when i run the script in a TS, i've moved the steps so it runs in the current OS and not in WinPE, this is the error i get:
[image]

texhex, 2018-03-13:
Please attach the logfile so I can have a look.

Joseahan, 2018-03-13:
Here you go:
[BiosSledgehammer.ps1.log-04.txt](https://github.com/texhex/BiosSledgehammer/files/1807651/BiosSledgehammer.ps1.log-04.txt)

texhex, 2018-03-13:
Thanks. I think we have changed the BitLocker Decryption function about six times already, it's really hard to get it right.

Based on the output and the error message I believe this is caused by a volume that is BitLocker enabled but has no drive letter which makes the function to go BOOM.

I just pushed 3.4.1 which should include the fix. Please download the repo with this link: https://github.com/texhex/BiosSledgehammer/archive/master.zip

Copy BiosSledgehammer.ps1 and try again

Joseahan, 2018-03-14:
Works like a charm now :)
Got another small issue though, when the TPM upgrade is in progress and the script exits with 3010 because it needs the restart to finish my, Task Sequence fails and reports 3010 as a failure, got any idea what it might be?

texhex, 2018-03-16:
Good to hear, thanks for your patience. 

Regarding the exit code: You need to tell SCCM/MDT that 3010 is an expected error code and does not indicate an error. This should be explained in the docs. 

I'm closing this issue now.

closed 2018-03-16 by texhex