texhex/BiosSledgehammer #46: HP EliteBook Folio 9470m

napzero, 2018-03-21:
I tested on our own laptops.

[BIOS-Settings.txt](https://github.com/texhex/BiosSledgehammer/files/1834532/BIOS-Settings.txt)
[BIOS-Update.txt](https://github.com/texhex/BiosSledgehammer/files/1834533/BIOS-Update.txt)
[ME-Update.txt](https://github.com/texhex/BiosSledgehammer/files/1834535/ME-Update.txt)
[folders.zip](https://github.com/texhex/BiosSledgehammer/files/1834563/folders.zip)

texhex, 2018-03-21:
@napzero I think that issue is missing some explanation. Is this about adding the 9470m to out-of-the-box models list?

napzero, 2018-03-21:
Ah yes, my bad. I'm contributing this, if you wish to include it!

thecatdidit, 2018-03-23:
@napzero EliteBook 9470m settings worked like a charm. Thanks!

I'll be posting EliteBook 9480m soon, hopefully.

thecatdidit, 2018-03-23:
@napzero  @texhex HP EliteBook Folio 9480m posted!

texhex, 2018-03-23:
@napzero Thanks, will check it in a minute.

texhex, 2018-03-23:
@napzero I checked the files, everything we need is in there, thank you!

Just one question: You have disabled both UEFI and Secure Boot in BIOS-Settings.txt:

```
#Boot Mode==UEFI Native (Without CSM)
#SecureBoot==Enable
```

Does this mean the 9470m does not support these settings or they are supported but you do not use them?

texhex, 2018-03-26:
@napzero No problem, then we'll keep this example as is. Version [3.4.1](https://github.com/texhex/BiosSledgehammer/releases/tag/v3.4.1) includes your changes, thanks for contributing!