texhex/BiosSledgehammer #48: HP EliteBook Folio 9480m

thecatdidit, 2018-03-23:
I posted this in an earlier entry that was closed. I suspect it had to do with the formatting (hyperlinks to HP Support, etc.).

At any rate - here is a working config for the HP EliteBook Folio 9480m. This includes:

- BIOS Update 1.42 (SP85010)
- Intel ME 9.5.62.3002 (SP82540)

[BIOS-Update.txt](https://github.com/texhex/BiosSledgehammer/files/1842809/BIOS-Update.txt)
[ME-Update.txt](https://github.com/texhex/BiosSledgehammer/files/1842810/ME-Update.txt)
[BIOS-Settings.txt](https://github.com/texhex/BiosSledgehammer/files/1842811/BIOS-Settings.txt)
[HP EliteBook Folio 9480m.zip](https://github.com/texhex/BiosSledgehammer/files/1842814/HP.EliteBook.Folio.9480m.zip)

texhex, 2018-03-23:
Thanks @thecatdidit (great username btw :), this looks fine to me and it will be added. 

However, you have disabled both UEFI and Secure Boot in BIOS-Settings.txt:

```
#Boot Mode==UEFI Native (Without CSM)
#SecureBoot==Enable
```

Does this mean the 9480m does not support these settings or they are supported but you do not use them? If possible both settings should be one to follow Microsoft best practice.

thecatdidit, 2018-03-23:
@texhex Unfortunately, in my IT menagerie, the screams for wall-to-wall Windows 10 were like fighting city hall. I did create a Task Sequence block that flipped the EliteBook 94x0 models before imaging began. However, the timeline didn't let me test enough for a proper sign-off. 

You are right - that is the best practice. I just kept the file in there so that the model would have the correct syntax.

texhex, 2018-03-24:
@thecatdidit Sorry, but I think the first sentence exceeds my knowledge of the English language (I'm no native speaker). 

Does this mean that both settings are supported by 9480m and would not be a problem to enable them out-of-the-box? But you hadn't had any time to test them thoroughly?

texhex, 2018-03-26:
@thecatdidit Version [3.4.1](https://github.com/texhex/BiosSledgehammer/releases/tag/v3.4.1) includes your changes, thanks for contributing!

closed 2018-03-27 by texhex

thecatdidit, 2018-03-27:
@texhex - Correct, sir. I didn't have time to test those settings, but they would be supported should someone change them accordingly.

fsarnakur, 2021-07-01:
> Thanks @thecatdidit (great username btw :), this looks fine to me and it will be added.
>
> However, you have disabled both UEFI and Secure Boot in BIOS-Settings.txt:
>
> ```
> #Boot Mode==UEFI Native [...]

> Thanks @thecatdidit (great username btw :), this looks fine to me and it will be added.
>
> However, you have disabled both UEFI and Secure Boot in BIOS-Settings.txt:
>
> ```
> #Boot Mode==UEFI Native [...]

After setting this my laptop shows, "No bootable image found. Notebook will be shutdown."
Any solution?

fsarnakur, 2021-07-01:
> Thanks @thecatdidit (great username btw :), this looks fine to me and it will be added.
>
> However, you have disabled both UEFI and Secure Boot in BIOS-Settings.txt:
>
> ```
> #Boot Mode==UEFI Native [...]