texhex/BiosSledgehammer #49: HP EliteBook 840 G4 (Updated with BIOS 1.15/ME 11.8.50.3425)

thecatdidit, 2018-03-23:
Tested on three EliteBook 840 G4 machines, each running Win10 1709 (FCU).

- BIOS 01.15 REV: A PASS: 1 (SP85231)
- Intel ME Firmware Component 11.8.50.3425 A 2 (SP82486)
- TPM 7.62 (SP81900)

[HP EliteBook 840 G4 (updated 23 Mar 2018).zip](https://github.com/texhex/BiosSledgehammer/files/1842855/HP.EliteBook.840.G4.updated.23.Mar.2018.zip)
[TPM-BIOS-Settings.txt](https://github.com/texhex/BiosSledgehammer/files/1842858/TPM-BIOS-Settings.txt)
[TPM-Update.txt](https://github.com/texhex/BiosSledgehammer/files/1842860/TPM-Update.txt)
[BIOS-Settings.txt](https://github.com/texhex/BiosSledgehammer/files/1842861/BIOS-Settings.txt)
[BIOS-Update.txt](https://github.com/texhex/BiosSledgehammer/files/1842862/BIOS-Update.txt)
[ME-Update.txt](https://github.com/texhex/BiosSledgehammer/files/1842863/ME-Update.txt)

texhex, 2018-03-23:
Thanks @thecatdidit. The changes you made is the new BIOS version and the new ME version, all others are not changed, correct?

thecatdidit, 2018-03-23:
@texhex That is correct. The config for the TPM 7.62 that you provided in release **3.4.0** is exactly the same. I can run a file hash check if you'd feel more comfortable.

thecatdidit, 2018-03-23:
Updated post re: details on TPM

texhex, 2018-03-24:
@thecatdidit Thanks and the hash check is not needed, I just wanted to make sure that I don't missed something. 

I will include the changes in a minute.

texhex, 2018-03-24:
Changes included, I will try to make a new release available tomorrow.

texhex, 2018-03-26:
@thecatdidit Version [3.4.1](https://github.com/texhex/BiosSledgehammer/releases/tag/v3.4.1) includes your changes, thanks for contributing! 

I'm closing this issue now.

closed 2018-03-26 by texhex