texhex/BiosSledgehammer #50: Bios Version - Longer then [Major.Minor.Build.Revision]

esoroach, 2018-04-06:
Newer HP's handle version 1.0.0.0.5

But those wont work with "function ConvertTo-Version()"
MPSXM.psm1

Is there a simple way to make longer Versions Compatible with [System.Version]?

texhex, 2018-04-06:
Could you please post the link to the BIOS with this version, right now I would suspect the -RespectLeadingZeros option of the above mentioned function to be the problem. In case a BIOS version already uses all parts, the function will go bonkers.

esoroach, 2018-04-06:
HP EliteBook 830 G5 
Q78 ver. 01.00.05 01/25/2018
But the script handels it like 1.0.0.0.5

So your comment makes sense;)

The following fix for [1.2.3.4.x] to merge at the end is not needed then.
But just to share would it ever happen:
_______________________________________________________________
try
 {
            # Handels 1.2.3.4.xxx and Merge [4.x] to [4x] so System.Version will accept the value
           if ($count = ($Text.Split(".") |  measure-object).count -ge 5)
            {
             $textbegin= $text | % { $_.split('.')[0..3] -join '.' }
             $textend =  $text | % { $_.split('.')[-1] -join '.' }
             $textcombine= $textbegin+$textend
             [version]$version=$textcombine
                
            }
           else
            {
            # Just handels 1.2.3.4
             [version]$version=$Text
            }
 }

texhex, 2018-04-06:
Thanks, let me do some tests, I think we can skip the entire -RespectLeadingZeros for those new BIOS versions.

texhex, 2018-04-06:
The change only needs to be done in the function ``ConvertTo-VersionFromBIOSVersion()`` as this is the only place BIOS versions are parsed. The entire part with -RespectLeadingZeros was only designed for the "old" version numbers e.g. 1.03. 

Given that HP now has adapted a system that looks like a real version number, the function is changed that it searches if the version contains only one dot (e.g. 1.03) and only use -RespectLeadingZeros in that case. In any other case, the version number is used as is. This code should also work if HP ever uses full versions like 1.00.03.256. 

Here's the code part, this is the current version of BiosSledgehammer.ps1. No release so far, I hope to have a first 4.0 Beta release tomorrow. 

https://github.com/texhex/BiosSledgehammer/blob/668f213bcfc455406731b3ec85638b6838a5ab9b/BiosSledgehammer.ps1#L1093-L1111

texhex, 2018-04-07:
Version [4.0.5](https://github.com/texhex/BiosSledgehammer/releases/tag/v4.0.5) (still a BETA) contains this change, please give it a try.

esoroach, 2018-04-07:
Your Fast! 
I'll Test it Monday. 

*made some other functions/improvements I like to share thoughts about.

texhex, 2018-04-13:
@esoroach Any updates on this please?

texhex, 2018-04-26:
@dubl3a Thanks for the confirmation. Given that I also tested it in the meantime, I think it's save to close this issue now.

closed 2018-04-26 by texhex