texhex/BiosSledgehammer #54: Add TPM Activation Policy for ProBook 4xx/6xx devices (TPM SLB 9670)

texhex, 2018-05-07:
@datagutten I would like to update the above mentions models with a TPM-BIOS-Settings.txt that includes the following line:

```
# No F1 prompt for TPM update
TPM Activation Policy==No prompts
```
I do not have any of these models, but would expect that if they feature the SLB9670 this setting to be supported. 

Would this be OK for you?

datagutten, 2018-05-08:
If the model has SLB9670 the setting should be valid. ProBook 640 G1 has SLB 9656, but the setting is also valid there. Upgrading SLB 9656 is another story.

texhex, 2018-05-10:
Thanks, I think I was able to update all. The first commit documented above should actually read "EliteDesk 8x0 G3: Added TPM BIOS..."...

Regarding the SLB 9656 (G1 and G2) models: Yes, this is first class cluster f***. The IFX update tool requires the TPM Owner Password if it isn't saved to the registry. We disallowed that storage since it was a GPO settings, so the basic procedure in this case  is:

* Suspend BitLocker
* Remove TPM protector from BitLocker
* Clear TPM and disable Auto Provisioning of TPM 
* Reboot, user must approve clear of TPM (using PPI)
* Take Ownership of TPM with known password
* Reboot
* If this worked, start IFX tool with /pass parameter and the known TPM Owner Password
* Reboot
* Check if TPM was updated successful
* If so, clear TPM again and enable Auto Provisioning of TPM (so the owner password is again unknown)
* Reboot, user must approve to clear TPM (using PPI)
* Check if TPM was owned by Windows and if so, add TPM protector and resume BitLocker again

We have received a working script to do all this last week, but it took the external contractor 8 weeks to code this. No chance that I include all this in BIOS Sledgehammer.

texhex, 2018-05-10:
Changes made, closing issue.

closed 2018-05-10 by texhex