texhex/BiosSledgehammer #56: Add SLB 9670 v7.63 firmware files

texhex, 2018-05-17:
HP has updated the TPM firmware files for the SLB 9670: https://ftp.hp.com/pub/softpaq/sp87001-87500/sp87492.html. The maximum version is now 7.63. 

The archive also includes some 4.x and 5.x firmware files where I have no idea to which models they apply. 

A first version has been added to master (Shared\TPM SLB 9670 - 7.63).

datagutten, 2018-05-18:
After correcting the firmware directory from Firmware to src in TPM-Update.txt i was able to upgrade a EliteBook 840 G3 from 7.62.3126.0 without problems.

texhex, 2018-05-18:
First an update on those 4.x and 5.x firmware that are also in this package.

To me, it looks like the package contains ALL TPM Firmware HP has, even the ones that are for different TPM chips. For example, the included firmware *TPM12_4.43.257.0_to_TPM20_5.62.3126.0.BIN* matches exactly the firmware noted on [HPSBHF03568 rev. 6 - Infineon TPM Security Update](https://support.hp.com/lt-en/document/c05792935) for the EliteBook 1020 G2, EliteBook 740 G2 and several other types. 

However, the page says that those device feature a SLB 96 **6** 0 and not SLB 96 **7** 0 which is a requirement the document for this download notes. 

Sometimes you really wonder who does the QA for this type of downloads over at HP...

texhex, 2018-05-18:
@datagutten Thanks for the test and yes, I forgot to change the path.

I have changed firmware\ to src\ now and will set it as default with the next commit.

texhex, 2018-05-18:
All done, closing issue.

closed 2018-05-18 by texhex