texhex/BiosSledgehammer #58: HP EliteBook 9470m - Updated Model Folder with BIOS F.70

thecatdidit, 2018-05-18:
HP released **BIOS Update F.70** for the EliteBook 9470m. I have updated the _Model_ folder to reflect this. The folder also contains all of the other files as they are. Make it easy for a commit.

References:
**[https://ftp.hp.com/pub/softpaq/sp87001-87500/sp87466.html](url)**
[**](https://ftp.hp.com/pub/softpaq/sp87001-87500/sp87466.exe)**
[HP EliteBook Folio 9470m.zip](https://github.com/texhex/BiosSledgehammer/files/2017365/HP.EliteBook.Folio.9470m.zip)

texhex, 2018-05-18:
@thecatdidit Thanks for the update! 

For the future, please only include files that have changed in the ZIP, this make integrating it for me a little easier.

Your BIOS-Update.txt was somewhat off I think. The BIOS version noted was still F.69, although this was about F.70. 

Also, your files got the the -p"@@PASSWORD_FILE@@" parameter removed, so anyone with a BIOS password can't use the file. I guess you removed it because you don't use a BIOS password, but that's not necessary. If BIOS Sledgehammer detects a @@PASSWORD_FILE@@ parameter and the current machine does not have a BIOS password set, the parameter is silently removed so it will work in any case.

texhex, 2018-05-18:
Files added, closing issue

closed 2018-05-18 by texhex

texhex, 2018-05-18:
@thecatdidit No problem, I'm glad you informed me about the new BIOS and that I only want the changes is because I'm a lazy sloth :)

But if you want to warm up with GitHub, look up pull requests. They make changes really easy, both for the creator and the integrator.

thecatdidit, 2018-05-23:
@texhex Will do. In fact, I have another model to include, so I'll check it out.