texhex/BiosSledgehammer #61: Create new TPM update method that supports automatic firmware file selection

texhex, 2018-05-18:
SoftPaq [#87492](https://ftp.hp.com/pub/softpaq/sp87001-87500/sp87492.html), which includes version 7.63 of the firmware for the SLB 9670 chip (see #56), supports a new parameter so the matching firmware file is automatically selected.  Excerpt from SP87492.pdf which is included with the above mentioned SoftPaq:

 >-a*TPM-Spec-Version*
> Auto select only compatible image files that have the specified TPM specversion. This option must be used with ‘-t’ or ‘-s’.Note: There is no space between -a and TPM spec version.Possible spec version values:1.22.0

This would solve the problem of firmware updates were only the build version changed (e.g. 6.41.197 vs 6.41.198) that BIOS Sledgehammer can not difference because Windows only returns Major.Minor (6.41 in this case) for the TPM firmware version. Also, it would make the TPM-Update.txt a lot shorter as all firmware version configuration entries could be removed. 

As the ``-s`` parameter expects the BIN files to be in the same folder:

> Automatic firmware selection with specified spec version looks for compatible
firmware files matching the spec version on the same folder as the program

We would just need to add one parameter that defines in which sub folder the firmware files are, to copy it together with the update utility to the same local execution folder.  

The command to update to TPM 2.0 afterwards would then just be:
```
C:\SomeTempFolder\TPMConfig64.exe -s -a2.0 -p"@@PASSWORD_FILE@@"
```

Other notes:
* The PDF does now also contain all return codes and their meaning, we should use that list to translate error codes to plain English and log them
* This will most likely be a breaking change
* This will not solve the problem if the latest version has two different firmwares (e.g. 7.63.3353 and 7.63.3144) because this could lead to a execution loop 

@datagutten Thoughts?

datagutten, 2018-05-19:
This would make it much easier to maintain. Return code checking would be useful when the upgrade fails.

texhex, 2018-05-20:
Thanks, so I will add this to my TODO list 

One request: Do you happen to have another G3 or G4 model where you could test the new -s -a2.0 command? 

I only have G5 devices at the office and they already have 7.63. I will try to get a G3/G4, but this might take some time.

datagutten, 2018-05-22:
I tested on a 640 G2, 840 G4 and a 430 G4 today and all worked, but it requires that VTd is disabled before starting the upgrade.

datagutten, 2018-05-22:
What about logging the changes made by TPM-BIOS-Settings.txt and reverting them after the upgrade?

texhex, 2018-05-22:
> I tested on a 640 G2, 840 G4 and a 430 G4 today and all worked, but it requires that VTd is disabled before starting the upgrade.

Good catch! I never noticed this because we turn off VTd by default. Will update the docs right away.

texhex, 2018-05-22:
I changed VTd in most TPM-BIOS-Settings.txt with the exception of *Models\HP ProDesk 400 G2* and
*Shared\HP ProBook 6x0 G2*. There are no example BIOS-Settings so I can be sure that setting exist.

Also, for *Shared\HP EliteDesk 8x0 G3*, Intel Software Guard Extensions (SGX) was found inside the BIOS setting but not in TPM-BIOS-Settings.txt, so I added it.

texhex, 2018-06-05:
@datagutten Sorry for the delay, but it was quite busy the last days. I reverted the change in TPM-BIOS-Settings.txt from ``VTd`` to ``VTx`` for all models. I would be great if you could provide some example BIOS Settings for the following models so we could provide a matching TPM-BIOS-Settings.txt for them as well:

* ProBook 640 G2
* ProBook 650 G2
* ProDesk 400 G2
* ProDesk 400 G3 DM

So far, no reply from my HP support, I guess they are busy getting the G5 out of the door.

texhex, 2018-06-27:
Thanks, the settings for ProDesk 400 G3 look fine, especially as VTx is enabled again (which will be turned off during the TPM update). BIOS settings for 640/650 G2 would be great and I think it's OK to assume that the settings from the 650 G2 can be "recycled" from 640 G2.

I will try to have the code for the new TPM update ready by the weekend (I was offline for two weeks).

texhex, 2018-07-04:
@datagutten Thanks a million! Your PR #65 also includes the BIOS settings for 640 G2 and 650 G2 so we have matching BIOS settings examples for all devices that are able to update the TPM. I will merge it in a second.

texhex, 2018-07-04:
Before heading to 5.0 with the new TPM command, I just release 4.0.7 with all your changes. Please see https://github.com/texhex/BiosSledgehammer/releases/tag/v4.0.7

texhex, 2018-07-08:
@datagutten The very first version of the new TPM update code is now online (as 5.0 BETA). 

It includes the code cleanup, a description of the return code of TPMConfig, the new arguments and an updated TPM-Config.txt in \Shared\TPM SLB 9670. What is missing is copying the \SRC folder because I still somewhat hope that the TPMCOnfig looks in that folder itself. A first version of the documentation is also online:  https://github.com/texhex/BiosSledgehammer#50-beta-documentaion-tpm-update

Except for the changes regarding Update-TPM, this is the same version as 4.0.7 so it's safe to test. Do you have any devices you could do a test run? We only receive G5s by now that all have 7.63 installed, so I can't test it. I do not know when the next batch of older devices comes in that I could use for testing.

datagutten, 2018-07-09:
I tested today and I need to copy TPMConfig64.exe to the src folder to make it work. It is strange that HP provides a package that does not work out of the box.
Before copying the file it failed with correct error handling. After copying it worked without problems.

texhex, 2018-07-09:
So much for my hope that maybe the HP devs are smart...

Thanks for test, and good to hear the error handling works as expected. Please give me some days to think about how to implement this "copy folder from source to source" in a way that it can be reused for other cases as well and without breaking anything.

Most likely, I will copy the \SRC folder to the base of the folder where TPMConfig64.exe is (the other way to what you did). Your method is clearly faster because the firmware files are so big, but I would like to keep the path inside command as close as possible to the source layout.

texhex, 2018-07-10:
The TPM-Update.txt does now support the setting `AdditionalFilesDirectory` that can be used to copy files from a subfolder of the source to the root of the update temp folder. This should do the trick, during my mocking tests, it worked. Please pull both the script and Shared\TPM SLB 9670\TPM-Update.txt and try it again, if your time permits. Now it should work "out of the box", without any copying necessary. 

If you have a better name for this special setting (AdditionalFilesDirectory), I'm open for suggestions.

texhex, 2018-07-19:
I got two older devices today and tested it. It worked as expected, so I think the code is finished. 

@datagutten If you could still give it a try, this would be appreciated.

texhex, 2018-07-23:
As @datagutten is on vacation and I was able to test this today on three more devices, I consider this code to be stable.

This change is live in [v5.0.6](https://github.com/texhex/BiosSledgehammer/releases), closing issue.

closed 2018-07-23 by texhex