texhex/BiosSledgehammer #66: prodesk 400 g3 shared and elitebook 840 g4 shared

devtestmagdevtest, 2018-07-05:
Prodesk 400 shared files
https://ftp.hp.com/pub/softpaq/sp87501-88000/sp87843.html
https://ftp.hp.com/pub/softpaq/sp87501-88000/sp87843.exe

elitebook 840 g4 shared files
https://ftp.hp.com/pub/softpaq/sp87501-88000/sp87849.html
https://ftp.hp.com/pub/softpaq/sp87501-88000/sp87849.exe

texhex, 2018-07-07:
At least some word about what this issue is about would be nice.

karlis-vitols, 2018-07-19:
If I understand him correctly, then he means the example .exe download script (StartExampleDownloads.bat). Some of the links right now are not working, because they are  pointing to the addresses mentioned above. Obviously, it breaks the download .script, but can be fixed by changing SPDownload.txt files for the affected models.
Do you need a list? 
There are some 4 or 5 models with this problem. What is funny, that ME FW version is not actually changed, its just that SP have different numbers now.

datagutten, 2018-07-19:
sp87434 is labeled "Intel Corporate Management Engine (ME) Firmware Component - Kaby Lake" with version "11.8.50.3470 REV: A PASS: 2", sp87849 is labeled "Intel Management Engine (ME) Firmware Component" with version "11.8.50.3470 REV: A PASS: 1".
I thought I had submitted this already, but it was only in my working directory and not committed.

texhex, 2018-07-19:
Sometimes you really wonder if there is somebody at HP Inc. that works in the "How to break things and surprise the customer" department....

@devtestmagdevtest @karlis-vitols Thanks, now I understand. 

@datagutten If you already have the files ready to go, I would appreciate a PR and then merge it.

Regarding the list @karlis-vitols: Thanks, that would be nice but please let's first wait if @datagutten creates a PR that will fix those. If after that still some things are broken, I would be happy for any help on this.

texhex, 2018-07-19:
@devtestmagdevtest The PR #68 from @datagutten should fix the issues you have found, so I will close this issue. Please feel free to reopen if something is missing.

@karlis-vitols If there are any other broken links, please open a new issue and we'll try to fix them. Thanks.

closed 2018-07-19 by texhex