texhex/BiosSledgehammer #70: HP ProDesk 400 G2.5 "Asset Tracking Number" not an option

dreary-ennui, 2018-07-19:
Having trouble with the HP ProDesk 400 G2.5 SFF models, looks like there is no "Asset Tracking Number" option, instead called "Asset Tag". Let me know how best to proceed, thanks!

texhex, 2018-07-20:
This is somewhat strange, giving that we support the "normal" ProDesk 400 G2, but the log clearly shows that the setting is now known.  Additionally, even if it would support this, no BIOS update can happen because the BIOS version is reported as "02.17" without any family or other information. This is also not common. 

Checking the log, is it possible that you tried to execute BIOS Sledgehammer during PE? If so, could you please run it verbose after PE and sent a new log so I can check if at least the BIOS version works after that. 

I will in the meantime think about a way to allow more than one setting for testing the BIOS password.

dreary-ennui, 2018-07-20:
Was executing in PE, yeah - I do the activateUEFI there to automate a legacy ->UEFI prior to striping down the OS. However, seeing the same result once in Windows, unfortunately :(. I feel like this model might be a bit of a black sheep for HP.
[BiosSledgehammer.ps1.log-01.txt](https://github.com/texhex/BiosSledgehammer/files/2214441/BiosSledgehammer.ps1.log-01.txt)

texhex, 2018-07-20:
@dreary-ennui Thanks for the update, let me think about methods to bring this black sheep back to the family : ). 

I have some ideas for the BIOS version parsing as well as as the Asset Tracking Number test. I need to make sure to implement this in a way and doesn't break existing machines. 

Hopefully, I have some news tomorrow.

texhex, 2018-07-21:
@dreary-ennui The necessary code changes are done.

* The BIOS data parsing will now handle devices, that do not report a BIOS family, and process the result as just the BIOS version
* Before testing the BIOS password files, BIOS Sledgehammer will check if the device support a setting named ""Asset Tracking Number" or "Asset Tag". What turns out to be supported will be used.

There is no release for this as of yet, so just download the current snapshot from https://github.com/texhex/BiosSledgehammer/archive/master.zip and copy `BiosSledgehammer.ps1` from the archive to the ProDesk 400. Give it a test run in Windows directly and if looks good, please try it from SCCM/MDT. 

If possible, I would like to have the verbose log files for both cases, just so I check if everything looks normal. I can't test this directly as we do not have any ProDesk 400 G2.5.

dreary-ennui, 2018-07-21:
Awesome, will check out early next week. Thanks so much! Loving the utility, btw.

texhex, 2018-07-21:
You're welcome, happy to hear you find it useful.

dreary-ennui, 2018-07-23:
Ok - good news, bad news. The good news is it looks like your modifications take care of the original issues, thank you much for that. The bad news is that it appears - at this point - that on this model there are some settings that are not configurable through the BCU? Getting a return code of 6 when trying to modify the value for Asset Tag even running the BCU directly. Attached, BIOS sledge logs and a get return. Will keep digging and let you know of any updates.
[400g2.5_settings.txt](https://github.com/texhex/BiosSledgehammer/files/2219845/400g2.5_settings.txt)
[BiosSledgehammer.ps1.log-03.txt](https://github.com/texhex/BiosSledgehammer/files/2219846/BiosSledgehammer.ps1.log-03.txt)

texhex, 2018-07-23:
Well, that's rather unfortunate. We need some setting that we can change to detect the correct password, so...

* Checking the previous release of this BIOS, there is no indication of a bug that indicates this to be a bug
* The BIOS settings export shows that this is actually the "MOD v02.17" release, so release 2.17 for the MOD BIOS family
* Checking [SP87861](https://ftp.hp.com/pub/softpaq/sp87501-88000/sp87861.exe), the BIOS family matches, so we have definitely the right BIOS for this machine (some models have more than one BIOS family)
* What is rather uncommon is the addition of `(ReadOnly)` inside your settings export, I have never seen this before:
```
Asset Tag (ReadOnly)
	2Uxx383xxC
Ownership Tag (ReadOnly)
	
Born on Date (ReadOnly)
	00/00/0000
```
* Searching for this, I found [this question](https://h30434.www3.hp.com/t5/Business-PCs-Workstations-and-Point-of-Sale-Systems/HP-BIOS-Configuration-Utility-BCU-how-to-set-Ownership-Tag/td-p/6491317) on the HP forums. Unfortunately, without answer.
* My first idea was to use Ownership Tag in case Asset Tag could not be used, but since it's also marked as "ReadOnly" I would think changing that will result in another ACCESS DENIED error.
* I can't see any "Lock Settings" in this BIOS as some modern BIOS version allow, so this can't be reason either. 

There are two things you could try:

1) Factory reset the BIOS, then try a settings export again just to make sure if this make the ReadOnly entries go away
2) Use the F10 BIOS to define a password, then try a setting export and see if ReadOnly has changed.

texhex, 2018-07-23:
I just released v5.0.6 (BETA) that includes the newest BCU version. You might want to give this a try. 

Please remember to copy both *BiosSledgehammer.ps1* and *MPSXM.psm1*

texhex, 2018-07-29:
@dreary-ennui Any updates on this?

dreary-ennui, 2018-07-29:
Sorry Michael - was in training all last week. Should have an update by Wednesday.

dreary-ennui, 2018-07-31:
More bad news, unfortunately. Tried your suggestions and the new version to same result. Also interesting - a new BIOS update was released just a month ago, version 2.23 which doesn't work with the HPBIOSUPDREC utility - says the BIOS version doesn't match expected family syntax 👍 . Updated to 2.23 using the SAFUWIN utility packaged in SP87861, which did update properly, but still with the (ReadOnly) on the ownership info stuff. I can go into the BIOS setup and use CTRL+A to get into an "edit" mode for the ownership info, but that doesn't help much here. I'm going to open a support ticket with HP regarding BCU access to these settings and see what they say - but not with bated breath.

texhex, 2018-07-31:
Thanks for the update, please let me know what HP says. If even their own BiosUpdRec utility fails, something is really not right there.

dreary-ennui, 2018-07-31:
Tell me about it - told ya it was the black sheep of the family :D

Warranties are starting to expire on our fleet so I might be able to eventually swing a “best effort” angle but I’m gonna keep trying in the meantime. Will keep you posted. Thanks much for your support.

texhex, 2018-09-11:
@dreary-ennui There was no update on this for the last three weeks, please let me know if there are any news from HP?

dreary-ennui, 2018-09-11:
Hey Michael - thanks for following up. Doesn’t look like we’re getting anywhere with this and we’re probably just going to cut our losses on this model seeing as how they’re coming off of warranty anyway. You can close this issue, sorry!

texhex, 2018-09-13:
OK, sad. Would have been great to get a response from HP for this. 

Everything I was able to gather in the meantime indicates that these (ReadOnly) attributes appear for some BIOS settings when the device is considered to be a customer device. But that's just an educated guess as long as HP does not confirm it.

closed 2018-09-13 by texhex