texhex/BiosSledgehammer #75: Script to generate SPDownload.txt file

napzero, 2018-09-10:
I'm not sure how you create your SPDownload files, but I'm sharing a small batch file I made to generate them. Perhaps it will be helpful.

[create-new_SPDownload_txt.zip](https://github.com/texhex/BiosSledgehammer/files/2367987/create-new_SPDownload_txt.zip)

texhex, 2018-09-11:
Thanks Matthew, but right now I'm a little bit stuck about the use for this. 

I think this would be more of a tool for a contributor than a normal user, right? I would expect normal users just download the SP, store the files and be done with it without touching SPDownload.txt. 

Or do you use it internal for your installation? If so, the use case would be interesting for me.

texhex, 2018-09-12:
Ah, now I understand. And yes, you are right - using the script that way make handling the SP download and extraction a lot easier. To be honest, I never thought about using it that way. 

Give me some days to think about it how to describe this and come up with a new name; ExampleDownload doesn't make sense when using that workflow.

napzero, 2018-09-12:
If I may make a name suggestion... SP-LoadAllMissing

texhex, 2018-09-15:
@napzero Why did you closed it?

napzero, 2018-09-15:
Not really sure, just figured it didn't need to stay open.

texhex, 2018-12-19:
@napzero Well, it was a little bit more than "some days" but I finally took the time to make the necessary changes. The download PS is now named "StartSoftPaqDownloads" and there is a new section inside readme that links to the issue and your batch file. 

Please have a look at https://github.com/texhex/BiosSledgehammer#adding-firmware-files and let me know what you think.

napzero, 2018-12-27:
Looks cool, thanks. I was thinking though... why is there a separate SPdownload.txt file for each model... wouldn't it be easier to maintain if all of them were listed in a single file like a database? Not a trivial question, I know. It would require a substantial rewrite, right?

texhex, 2018-12-27:
Thanks, so I will leave the text as is.

Regarding your question: Yes, this would for one require a rewrite but at the same time a central "SoftPaq Catalog" file/database would not make any sense as BIOS Sledgehammer is designed around the principle that there is one folder per model that holds all required files and settings. Creating a central files for SoftPaq would violate this principle.

In case several models share the same BIOS download, they will also share the same BIOS settings so you can create a \Shared folder for those and have everything in one folder again.

closed 2018-12-28 by napzero