texhex/BiosSledgehammer #76: Disable TXT require shutdown?

JohFro, 2018-09-14:
Hi, I’m trying to do TPM firmware upgrade using SCCM task sequence on HP EliteBook Folio 1040 G3 with bios version 01.33. I have trouble disabling TXT before the TMP firmware upgrade and the task sequence fails.
When running the BiosConfigUtility64.exe /set:"1040G3.txt" /cspwdfile:"pwd.bin"  from inside Windows I get the message:
<REBOOT msg=”Reboot required to perform BIOS settings update. **May need to shutdown if settings not take effect**.”
If I reboot the computer the TXT setting is not disabled, if I do a shutdown the TXT setting gets disabled, does someone have any suggestion how to disable TXT (and VTx) without doing a shutdown of the computer?

From settings file:
Trusted Execution Technology (TXT)
	*Disable
	Enable
Virtualization Technology (VTx)
	*Disable
	Enable

texhex, 2018-09-14:
Where do you use BIOS Sledgehammer in this process?

texhex, 2018-09-14:
OK, understood. For some settings to apply the HP BIOS requires a restart because the change happens only during POST and not while the BIOS is already started. No way around it. 

I will close this issue now because it's out of scope. Please feel free to open a new one in case you have a question about BIOS Sledgehammer.

closed 2018-09-14 by texhex