texhex/BiosSledgehammer #78: BIOS update needs BIOS-Update-Settings.txt

merlinfrombelgium, 2018-10-16:
For reference, I encountered this with HP ProBook 650 G1/G2 models.

If the BIOS setting '**Update System BIOS**' is set to **Disable**, the computer reboots after SledgeHammer pass 1, the BIOS image is written to the system, computer reboots again and user is prompted for BIOS admin password. This of course breaks automation.

Since applying the BIOS-Settings.txt file has the least priority, we need something else to set certain BIOS settings before the BIOS update.

I would suggest to add a settings file for BIOS update, similar to TPM-BIOS-Settings.txt. Name it BIOS-Update-Settings.txt or something.

texhex, 2018-10-18:
Thanks for the report and you are right, if this setting is set we have an infinite loop. I like your idea with the additional file but can you please try if enabling this value and THEN applying the BIOS update (both in Windows) will actually work? 

If not, we would plan for an reset cycle or find a new method that checks the value (and others) and change them if required.

merlinfrombelgium, 2018-10-22:
I had to read that a couple of times before understanding what it is you want me to test. After 2 coffees, I got it :)

Will validate and report back. Stand by.

texhex, 2018-10-22:
I'm sorry, this was poorly written. But good to know that you were able to transcribe it using a brew of magical beans :)

What I meant was that we need to check if this setting has an effect in case we are setting in AND updating the BIOS at the same POST phase.

I was thinking about this:
- Set `Update System BIOS` to `Disable`
- Confirm, boot Windows and restart right away (to make sure it's activated)
- In Windows, use `BCU.exe` to set this value to `Enabled` (no restart)
- Let the script update the BIOS
- Restart the machine and see what happens during POST: Does the BIOS update happen right away, without any password entry, or do you need to enter the password again?

texhex, 2018-11-01:
@merlinfrombelgium Any updates on the test run?

merlinfrombelgium, 2018-11-01:
None so far. Haven't gotten round to it. Next Monday should be better!

texhex, 2018-11-01:
Alright, thanks.

merlinfrombelgium, 2018-11-08:
Haven't forgotten about this! I keep pushing it back though, sorry. Will definitely get back to this.

texhex, 2018-11-08:
Thanks. I would be great to get a feedback next week so I could include this into the next release.

texhex, 2018-12-19:
Assuming this issue to be abandoned and closing it. 

Please feel free to reopen it if you have the results and we could give it a test run.

texhex, 2018-12-19:
No problem, happy to reopen it. And yes, I still think it would be a great addition and happy to work with you on it.

merlinfrombelgium, 2018-12-20:
So the issue was reproduced on the 650 G1, but not on the 650 G2.

Your test method indeed allowed to update the BIOS without prompt. I didn't need to reboot in between updating the `Update System BIOS` to `Enable` and running the SH script to update the BIOS.

On the G2, turns out this setting isn't present. Must have confused myself somewhere while figuring it out.

The conclusion is that it would certainly be beneficial to have certain settings in the BIOS checked and remediated before performing the BIOS update. I can free up some time during the holiday period to work in the script with you, if that's how you'd like to proceed.

merlinfrombelgium, 2018-12-21:
Cool, let me know where I can help.

As a different note, I've been using your script beyond the scope you probably intended it to be used. Like in OSD (WinPE) and in BitLocker implementation. And I've had a thought. It would be cool to use the script with a parameter for specific updates: BIOS update, TPM, ME … The idea is to override/ignore the update order as scripted. Again, happy to help figure out how to. Is this something to open a new issue for? Not sure on the preferred approach.

texhex, 2018-12-22:
Please open a new issue for that, it would be interesting to know in which way you alter the use-case of it. Especially, I would like to know what you mean with "specific updates" because the order BIOS Sledgehammer uses should be correct, but I might be wrong there.

merlinfrombelgium, 2018-12-28:
Sweet, thanks for the work you put into this! Will try this out in week 2 of 2019 when I get back to work. Happy holidays dude!

texhex, 2018-12-28:
You're welcome, and the same to you!

I reopened the issue because I want it to have open until the code was verified by you and works as expected. It's not a problem if it stays open for two or more weeks. This isn't your internal ticket system where the supervisor keeps and eye on the number of open tickets 😉

merlinfrombelgium, 2018-12-28:
I actually clicked close and comment by mistake and then thought it didn't really matter :)

Yeah thanks for not breathing down my neck. I got plenty of people doing that already on the job ;)

texhex, 2019-01-15:
@merlinfrombelgium Did you had a chance to test this already?

merlinfrombelgium, 2019-01-16:
Not yet Michael. Next Monday/Tuesday will be next opportunity.

texhex, 2019-01-23:
@merlinfrombelgium ?

texhex, 2019-02-27:
@merlinfrombelgium Any updates?

merlinfrombelgium, 2019-03-01:
@texhex no 🙁

merlinfrombelgium, 2019-03-04:
This week, I have 3 days on-site at that particular customer. I'm making a solemn promise to you now that I will test this! ✌🏻

texhex, 2019-03-07:
Thanks, let me know if it works as expected.

merlinfrombelgium, 2019-03-07:
So... I tested the bios-update-settings on a ProBook 650 G1 today. Luck had it, HP pushed out a new BIOS version. Or you could say, it took me so long, they got bored and decided to update this old machine for fun 😜

Works as described! Used your template, added a value for Ownership Tag, just as an extra check. Watched the log live, as the script ran to verify settings were applied. BIOS update was next, script exited with 3010. After a manual reboot, BIOS update applied without human intervention and the Ownership Tag was set as well.

Thumbs up 👍

texhex, 2019-03-31:
Sorry for the long delay!

Thanks for the verification and good to hear it works as expected. I will keep this open until I have 6.0 ready. I'm not quite sure if I keep the current file name, the title of it is a little bit misleading. Just as a heads up when 6.0 comes out and your current configuration stops working.

texhex, 2019-04-14:
@merlinfrombelgium For your information, I will change the filename of this file to ``BIOS-Update-BIOS-Settings.txt``, simply to avoid that it sounds like it includes settings for the BIOS update process. I will provide an batch file later on that can rename all your existing files in one go.

texhex, 2019-04-14:
Here is the file to rename your existing file in one go. Just copy to the root of the BIOS Sledgehammer installation (where BiosSledeghammer.ps1 is), rename the extension from TXT to BAT and start it. 

This will rename all existing BIOS-Update-Settings.txt files to the new name. 

[zRenameConfigFilesForV6.txt](https://github.com/texhex/BiosSledgehammer/files/3077610/zRenameConfigFilesForV6.txt)

merlinfrombelgium, 2019-04-14:
Cool. Will be a while probably before I use the script again.
Will you attend MMSMOA next month? If so, let's have a beer or two at the Firelake!

texhex, 2019-04-14:
Thanks for the offer, but as I first needed to google MMSMOA and I don't plan to fly to the US, I won't attend. But if you want to do me favor, point some more people to BIOS Sledgehammer. I'm happy if it's used by more people. 

As a side note: That somebody from BELGIUM really offers somebody else American "beer" is somewhat... strange. Don't know if this was intended as insult ;-).

merlinfrombelgium, 2019-04-14:
I see your point. But they do have Belgian beer there :) And some very nice locally brewed beer as well.

I do suggest my customers and peers to use your script as it has proven its worth. I think there is still some undeserved trust issue around community tools. But the way you have documented and how you support your script, deserves to be treated as a proper production ready tool. I mean that, and that's how I sell the idea to others. I have in fact included your tool in a proposal for a new staging process at a big customer. With full credit to your name of course. I'll let you know when we get the deal signed.

About MMS, I assumed you would know about it, as you are clearly familiar with managing PCs. In any case, you would definitely be in your element there. It is the best event for technical people to meet and learn. And it's all about community and finding better ways of doing what we do. You should consider it for next year! Let me know if a recommendation to your boss or whoever decides about the budget, would help. I would gladly help to convince them so you could join us.

texhex, 2019-04-24:
Thanks, that's much appreciated and great to hear! Please let me know if the customer accepted it, I think it would be one of those email I would print out.

MMS: Thanks, I see. Maybe next year; right now no business trips, that are not considered business critical, are allowed, so no chance. Have fun there!

As a side note: I should be able to release 6.0 BETA 1 end of this week or start of next week. If you have some spare devices, it would be great to give it a try.

texhex, 2019-05-05:
A week later than announced, but now 6.0.1 BETA is available from [/releases](https://github.com/texhex/BiosSledgehammer/releases) and includes the change and the batch file I already attached here. It would be great if you could give it a test run. 

I'm closing this issue now, if you detect anything wrong with the new version, please feel free to reopen it.

closed 2019-05-05 by texhex