texhex/BiosSledgehammer #79: Running BiosSledgehammer.ps1 from sccm task sequence

hanfelt, 2018-10-18:
Hi there when trying to run Sledgehammer from task sequence it always fails for me. Tried making package, run command, run powershell about everything you can think of. If i hit F8 and run it manually from command i get this error:
Start-TranscriptIfSupported: Cannot validate argument on parameter 'Path'. The argument is null or empty. Provide an argument that is not null or empty, and then try the command again.
At MPSXM.psm1:561 char:9
Start-TranscriptIfSupported - Path $logPath -Name $logName -Ne ....

texhex, 2018-10-18:
This is odd. Can you please enter the command ```$PSVerionTable``` inside a PowerShell command and post the results?

Also, just to be sure: You start this once the OS has started for the first time, not within Windows PE?

hanfelt, 2018-10-18:
PSVersion: 5.1.17134.1
PSEdition: Desktop
PSComatibleVersions: 1.0 2.0 3.0 4.0 ....
BuildVersion: 10.0.17134.1
CLRVersion: 4.0.30319.42000
WSManStackVersion: 3.0
PSRemotingProtocolVersion: 2.3
SerializationVersion: 1.1.0.1

hanfelt, 2018-10-18:
I commented out the line 561, 565 and 713 in MPSXM.psm1

561 # Start-TranscriptIfSupported -Path $logPath -Name $logName -NewLog
565 # Start-TranscriptIfSupported -Path $logPath -Name $logName
713 # Stop-Transcript
At least it runs now ;)

texhex, 2018-10-18:
OK, my first guess was that in your image you used an older version of PowerShell, which is clearly not the case. Given that you are using 5.1, it is really, really strange because we had a similar bug in the past with the logging: issue #40.

I think the error starts here in the main script (line 55+):
```
#log the output
Start-TranscriptTaskSequence -NewLog
```

However, after the above mentioned bug was identified, I tried my best to work around it. Somehow this does not work for your system, so it would be great if you copy back the original MPSXM.psm1, and start the script with the verbose switch (`.\BiosSledgehammer.ps1 -Verbose`), and attach the output (a screenshot will do, no text file required) to this issue. 

Hopefully the additional output there can help me to understand what is not working.

datagutten, 2018-10-18:
I have the same problem, it can not resolve $env:TEMP, so I have set it static to c:\windows\temp

hanfelt, 2018-10-18:
I will do that tomorrow at work and then post the result, thanks for all the help.

texhex, 2018-10-18:
Hmm, the problem with $env:TEMP again? I thought I had solved this already, but it seems not to be the case. @hanfelt After the verbose run, please also check what PS returns for `$env:TEMP`.

texhex, 2018-10-23:
@hanfelt Any updates?

hanfelt, 2018-10-24:
@texhex I tried yesterday and put the original file back but forgot to untick continue on error in the task sequence sorry ;) Then we got alot of other work to do so i couldnt continue, im off work today and might be home tomorrow as well (sick kids). I will do this first thing in the morning when im back. sorry for this delay

texhex, 2018-10-24:
No problem, take your time. I just wanted to make sure this isn't stalled. All the best for your kids!

hanfelt, 2018-10-26:
$env:TEMP = "C:\Windows\TEMP"
[image]
[image]

texhex, 2018-10-27:
Thank you, and that's really the worst possible outcome for a log function to cause the entire script to die on the spot. 

Let me check tomorrow why logPath could be empty and how to workaround it.

texhex, 2018-10-28:
I updated the script and hopefully this bug is now fixed. 

I have the possible explanations what is happening here:
- The variable `LogPath` in your SCCM environment is not set, and BIOS Sledgehammer simply used it as is (even if it's empty). This of course will not work, so we are defaulting to C:\Windows\Temp now. 
- Somehow, this variable is set but PowerShell can not retrieve the value so it's also an empty string. Also defaults to C:\Windows\Temp now.
- For whatever reason, my script can not determine the script filename. But it requires this to know how to name the log. Now try a second method is tried in this case.

Finally, the Start-Transscript*() functions already had verbose output to know what they are doing, but this was not used by BIOS Sledgehammer when starting it with the `-Verbose` parameter. This is now the case, so these function will also add verbose output.

Please do the following:
- Download the current version of this repo: https://github.com/texhex/BiosSledgehammer/archive/master.zip
- Extract `BiosSledgehammer.ps1` and `MPSXM.psm1` and copy them to your installation
- Retry the run

I really hope it will work this time.

texhex, 2018-10-28:
@datagutten Your issue with $env:TEMP will be checked once the issue for @hanfelt is fixed. If this new version works, it would be great if you could execute it on your system and attach the verbose log to a new issue.

hanfelt, 2018-10-29:
@texhex Will try this out tomorrow and get back to you with the result.

texhex, 2018-10-31:
Thanks for the update and sorry to hear that it still not work. A screenshot of the lower part (below the importing functions stuff) where the "real" verbose message begin might help me to understand why it still fails.

hanfelt, 2018-11-01:
Seems like it works i tried running it manually and it runs fine. It writes logs to C:\Windows\Temp as well

texhex, 2018-11-01:
Very good, but when run manually, the entire code that tries to read the data from SCCM is skipped, so there might still be an issue (just saying). If possible, it would be great if you could give it a test run in SCCM, as you said it didn't run OK for your colleagues.

hanfelt, 2018-11-01:
Still having to battle the other issue im really sorry for this ;) I just did i quick run today and thought it might be something else bugging out in the task sequence i did hit f8 and ran it from there and it worked. I didnt have time to do more today. Will look into this on monday next time Sorry!

texhex, 2018-11-01:
No problem, thanks for the update.

hanfelt, 2018-11-16:
Sorry for the late update @texhex , but i think everything works as it should, thanks alot for your help on this issue. Have a great weekend.

texhex, 2018-11-19:
You are welcome, thanks for the update. Here's hope that this will be the last time we need to fix an error in logging.

I will close the issue now.

closed 2018-11-19 by texhex