texhex/BiosSledgehammer #80: New HPFIRMWAREUPDREC tool with new HP models

merlinfrombelgium, 2018-11-23:
Just read this blog about a new executable to update BIOS on newer HP models. Seems there is no official documentation yet, so this post might help: https://www.systemcenterdudes.com/how-to-update-hp-bios-using-latest-hpfirmwareupdrec-with-sccm/
I've had trouble with TPM firmware updates when the HP_TOOLS partition wasn't available. That experience might come in handy as that partition now seems to be a dependency for the BIOS update as well. Maybe we should look into creating that partition as part of the script. Will be happy to help!

texhex, 2018-11-23:
To be honest, I somewhat doubt that. 

We are using ``HPFirmwareUpdRec.exe`` for the ProDesk 600 G4, EliteBook 8x0 G5 and Z2 G4 and I'm not aware of any issues that those updates will fail if the HP_TOOLS partition does not exist. Either the tools create them automatically, or this is something else (but I can't think of anything). I would need to check with the machines in the office, but as far as I remember, none of them had or has an HP_TOOLS partition. Normally, modern BIOS versions of HP have enough build in storage to fully store the new BIOS blob beside the current BIOS and then simply do the install/switch on next boot. 

For the TPM update: I would need the verbose output of BIOS Sledgehammer when this issues occurred. I have so far not seen any TPM update fail, so it would be interesting what is going on there.

And as side note: I would be nice if you could update #78 with the results of your test run :-)

texhex, 2018-12-19:
I checked the devices at the office, and none of them had the HP_TOOLS partition. However, I think I remember we once had an report about this: The explanation was that, when the BIOS update EXE is run in PE mode, it requires this partition to work. This might be the reason why your link says the partition is required. 

Since several HP tools are not supported in PE mode, and we also define that PE is not supported, I think it's safe to ignore this. I'm closing this issue.

closed 2018-12-19 by texhex

merlinfrombelgium, 2018-12-19:
Hi again,

If the requirement is hard linked to WinPE, than I agree with you on closing the issue, as you indeed state that WinPE is not supported.

For my OSD Task Sequences, I now create the HP_TOOLS partition as part of the 'Format and Partition Disk' step.