texhex/BiosSledgehammer #82: IFXTPMUpdate_TPM12_v0443.com

bitslacker, 2019-03-25:
This isn't really an issue; just a question:

Does BiosSledgehammer support the IFXTPMUpdate_TPM12_v0443.com program for updating TPM on the HP EliteBook Folio 1040 G2?

BiosSledgehammer is amazing by the way!

texhex, 2019-03-31:
Thanks, glad to hear you like it.

If I'm not mistaken that file is for updating TPM chips on models that do not support "Update through BIOS"? If so, I'm sorry this is not supported. 

When HP release the advisory I made a short plan what all needs to be done to support this and that would require a project on it's own. Starting with that update requiring the TPM Owner Password which normally is set once and then thrown away by Windows. So first, we would need to pause all applications that require the TPM chip, set a registry parameter that stops Windows from automatically trying to take ownership and reset it. When this works, we would need to take the ownership ourselves, reset the machine again and only then, maybe, we can update the TPM chip. 

That's the plan if everything works, but as you can imagine that are hundreds of possibilities this could go haywire...

bitslacker, 2019-04-01:
That's right and you pretty much described all the steps I have to go through to get it updated. I'm definitely glad they're not all like this!

closed 2019-04-01 by bitslacker