texhex/BiosSledgehammer #90: Ampersand in BIOS Settings

ryanbley3, 2019-04-19:
Trying to set BIOS setting: 
`Password prompt on F9 F11 & F12 == Enable`

Looks like the ampersand is escaping in XML. Might need to add a check for & character in BIOS Settings and inster the correct value.

ryanbley3, 2019-04-19:
It is the return message from the BCU it contains the & character and is breaking the reading of the BCU return information:
`VERBOSE: === BCU Result ===============================
VERBOSE: <BIOSCONFIG Version="" Computername="Removed" Date="2019/04/19" Time="14:42:57" UTC="-4">
	<SUCCESS msg="Successfully read password from file" />
	<SETTING changeStatus="skip" name="Password prompt on F9 F11 & F12" returnCode="18">
		<OLDVALUE><![CDATA[Enable]]></OLDVALUE>
		<VALUE><![CDATA[Enable]]></VALUE>
	</SETTING>
	<SUCCESS msg="No errors occurred" />
	<Information msg="BCU return value" real="0" translated="0" />
</BIOSCONFIG>
VERBOSE: ==============================================
PS>TerminatingError(): "Error trying to parse BCU output: Cannot convert value "<BIOSCONFIG Version="" Computername="Removed" Date="2019/04/19" Time="14:42:57" UTC="-4">
	<SUCCESS msg="Successfully read password from file" />
	<SETTING changeStatus="skip" name="Password prompt on F9 F11 & F12" returnCode="18">
		<OLDVALUE><![CDATA[Enable]]></OLDVALUE>
		<VALUE><![CDATA[Enable]]></VALUE>
	</SETTING>
	<SUCCESS msg="No errors occurred" />
	<Information msg="BCU return value" real="0" translated="0" />
</BIOSCONFIG>
" to type "System.Xml.XmlDocument". Error: "An error occurred while parsing EntityName. Line 3, position 64.""`

texhex, 2019-04-24:
Thanks for the report!

As you wrote, the error is clearly within the BCU tool, hence the best chance would be to open a case with HP so they can fix it. However, when checking which models support this setting, the only ones I found was ProDesk 600 G1, Z420 and a Z440. I have no idea if those devices are not already declared End-of-life by HP so they could simply deny the change in BCU. 

The only idea I had to get this working would be to add a "hack" that checks the output from BCU and replaces

``name="Password prompt on F9 F11 & F12"`` 
with
``name="Password prompt on F9 F11 and F12"`` 

How urgent is that you get this supported, is this a critical road block for you?

ryanbley3, 2019-04-26:
I just ran a -replace on the bcu results.

texhex, 2019-05-02:
Yes, at the end this will be likely the easiest method. I will implement this and include it with 6.0.

texhex, 2019-05-05:
Code is live, 6.0.1 BETA will be out in about 30 minutes, it would be great if you could test it if this fixed the bug for you. If not, please feel free to reopen this issue.

closed 2019-05-05 by texhex