texhex/BiosSledgehammer #91: Issues getting HP EliteDesk 800 G1 SFF to boot after applying BIOS settings

GregoryMachin, 2019-05-16:
Hi, 

Firstly thanks for a great tool. 

Is there a forum for users to discusses the BiosSledgehammer and other related topics, BIOS configurations and Firmware etc ? 

I'm posting this here as I'm not having much luck with the resources I have reached out to thus far.  If you feel this is not appropriate feel free to close this ticket. 

We are switching from Win7 to Win10, this is being done using SCCM task sequence (OSD) and is effectively a format and reload. I have setup BiosSledgehammer to reconfigure the BIOS as part of the initial steps. As part of re-configuring the BIOS we are switching to, UEFI. 

The Model I'm having the most trouble with is the HP EliteDesk 800 G1 SFF, how ever I configure it I get "No option to boot to", when I look in the BIOS there is no HDD listed under the UEFI boot order, but if I run the deployment multiple times it suddenly works. 

This is my current configuration : 

###Boot###
SATA Emulation==AHCI
EFI Boot Order==USB Floppy/CD,USB Hard Drive,ATAPI CD-ROM Drive,Network Controller IPv4,Network Controller IPv6,Other
PXE Option ROMs==UEFI Only
Storage Option ROMs==UEFI Only
Video Option ROMs==UEFI Only

Secure Boot==Enable

Legacy Support==Disable
Legacy Boot Vector Re-entry==Disable

OS management of Embedded Security Device==Enable
Reset of Embedded Security Device through OS==Enable
Tpm PPI policy changed by OS allowed==Enable
Tpm No PPI provisioning==Enable
Tpm No PPI maintenance==Enable

###CPU and Performance###
Turbo Mode==Enable
Hyper-Threading==Enable
Multi-processor==Enable
Virtualization Technology (VTx)==Enable
Virtualization Technology Directed I/O (VTd)==Enable
USB3.0 BIOS Driver Support==Enable

#BIOS 
Asset Tracking Number == @@COMPUTERNAME@@
Automatic BIOS Update Setting==Disable

Are there any errors ? 

My task sequence for apply the BIOS settings: 

- Create a MBR partition , so that the BiosSledgehammer has somewhere to be saved to. 
- Run BiosSledgehammer
- Reboot into WinPE
- Run BiosSledgehammer
- Reboot into WinPE
- Create UEFI Partitions
- Apply Operating System.
- Apply Drivers
- Reboot to Windows 10 < this is where the "No option to boot to" comes up
> continue TS

Are there any issues with this configuration?

Regards
G

texhex, 2019-05-20:
To my knowledge, there is no forum especially for BIOS Sledgehammer and to be honest, this wouldn't make a lot of sense to me. It's just one of the tools you are using to stage a PC, so I would think a general SCCM/Deployment forum makes more sense. 

Regarding your issue I have no idea where this comes from. The result of your changes should be deterministic, either it works all the time or never. 

There are two points here that might be worth a second look:

a) The BIOS calls the boot order "EFI", not UEFI. This could indicate that this device is not fully-UEFI compliant so maybe there is an BIOS update required before you try to boot first into Windows 10.

b) If I understood your order correctly, the device will have at the end an MBR and UEFI partitions. Maybe the BIOS gets confused by this when it tries to read which OS to load from the EFI partion (if the device has one). It would be interesting what BCDedit say. Or even better, give Visual BCD Editor (https://www.boyans.net/) a try and check if you see anything "strange"

texhex, 2019-06-25:
@GregoryMachin Were you able to find out anything about this issue in the meantime?

closed 2019-07-16 by texhex