texhex/BiosSledgehammer #92: Issues updating HP Elite Compaq 8300 BIOS

GregoryMachin, 2019-10-29:
Hi,

I'm struggelling to get the BIOS to update on the HP Elite Compaq 8300, HP Elite 800 G1. In both cases the BIOS update utility fails with  "Error Code  0x00000113  INCORRECT_PASSWORD". The BiosSledgehammer script confirms that the password file works, as does manual login, so I'm a bit lost, I haven't found anything useful on google. So I'm hoping I'm not the only one that has experienced this issue and someone can suggest an answer.  
 
Here is the BiosSledgehammer log:

**********************
Windows PowerShell transcript start
Start time: 20191029150156
Username: WAIKATO\adwmaching
RunAs User: WAIKATO\adwmaching
Configuration Name: 
Machine: computername (Microsoft Windows NT 10.0.17763.0)
Host Application: C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -ExecutionPolicy Bypass -File c:\Temp\BiosSledgehammer-Firmware Upgrade 10\BiosSledgehammer.ps1 -WaitAtEnd
Process ID: 1812
PSVersion: 5.1.17763.592
PSEdition: Desktop
PSCompatibleVersions: 1.0, 2.0, 3.0, 4.0, 5.0, 5.1.17763.592
BuildVersion: 10.0.17763.592
CLRVersion: 4.0.30319.42000
WSManStackVersion: 3.0
PSRemotingProtocolVersion: 2.3
SerializationVersion: 1.1.0.1
**********************
Transcript started, output file is C:\WINDOWS\temp\BiosSledgehammer.ps1.log-06.txt

            _
    jgs   ./ |   
         /  /    BIOS Sledgehammer Version 5.2.0
       /'  /     Copyright (c) 2015-2018 Michael 'Tex' Hex
      /   /      
     /    \      https://github.com/texhex/BiosSledgehammer
    |      `\   
    |        |                                ___________________
    |        |___________________...-------'''- - -  =- - =  - = `.
   /|        |                   \-  =  = -  -= - =  - =-   =  - =|
  ( |        |                    |= -= - = - = - = - =--= = - = =|
   \|        |___________________/- = - -= =_- =_-=_- -=_=-=_=_= -|
    |        |                   ```-------...___________________.'
    |________|   
      \    /     This is *NOT* sponsored/endorsed by HP or Intel.  
      |    |     This is *NOT* an official HP or Intel tool.
    ,-'    `-,   
    |        |   Use at your own risk.
    `--------'

 ####################################################################
 ##  THIS SCRIPT MIGHT ALTER YOUR FIRMWARE AND/OR BIOS SETTINGS    ##
 ##                                                                ##
 ##  You have 15 seconds to press CTRL+C to stop it.               ##
 ####################################################################

Verifying environment...
  Success
Verifying BIOS Configuration Utility (BCU) can communicate with BIOS.
  Trying to read Universally Unique Identifier (UUID)...
  Success
Collecting system information...

  Name.........: computername
  Model........: HP Compaq Elite 8300 USDT
  SKU..........: QV997AV
  BIOS (Raw)...: K01 v02.05
  BIOS Family..: K01
  BIOS Version.: v02.05  (Numerical: 2.0.5)
  TPM Vendor...: 1229346816
  TPM Firmware.: 3.17  (Numerical: 3.17)
  TPM Spec.....: 1.2  (Numerical: 1.2)

***** Locate Model Folder ***************************************
[... 83 lines cut ...]
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
at Update-BiosFirmware, C:\Temp\BiosSledgehammer-Firmware Upgrade 10\BiosSledgehammer.ps1: line 1623
at <ScriptBlock>, C:\Temp\BiosSledgehammer-Firmware Upgrade 10\BiosSledgehammer.ps1: line 3021
    + CategoryInfo          : OperationStopped: (:) [Write-Error], WriteErrorException
    + FullyQualifiedErrorId : Microsoft.PowerShell.Commands.WriteErrorException,BiosSledgehammer.ps1
C:\Temp\BiosSledgehammer-Firmware Upgrade 10\BiosSledgehammer.ps1 : BIOS update failed, update program returned code
259
At C:\Temp\BiosSledgehammer-Firmware Upgrade 10\BiosSledgehammer.ps1:1623 char:29
+ ...             throw "BIOS update failed, update program returned code $ ...
+                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
at Update-BiosFirmware, C:\Temp\BiosSledgehammer-Firmware Upgrade 10\BiosSledgehammer.ps1: line 1623
at <ScriptBlock>, C:\Temp\BiosSledgehammer-Firmware Upgrade 10\BiosSledgehammer.ps1: line 3021
    + CategoryInfo          : OperationStopped: (:) [Write-Error], WriteErrorException
    + FullyQualifiedErrorId : Microsoft.PowerShell.Commands.WriteErrorException,BiosSledgehammer.ps1

BIOS Sledgehammer finished, return code 666.
Thank you, please come again!
Waiting 30 seconds...
**********************
Windows PowerShell transcript end
End time: 20191029150321
**********************

Thank you.
G

texhex, 2019-10-30:
This is indeed very strange, given that the BIOS value change with the exact same password worked. 

I think the message does not mean "Password incorrect" but instead "Incorrect update path". For the Elite 8300, we seem to have the same topic as with the Pro 8300: it requires a [Two Step BIOS Update](https://github.com/texhex/BiosSledgehammer#two-step-bios-update-process) to move to 3.x.

Please first try flashing [v2.99](https://ftp.hp.com/pub/softpaq/sp73001-73500/sp73099.exe) (you can see this release by clicking "Previous releases" for BIOS v03.08), then update to 3.08.

texhex, 2020-02-02:
@GregoryMachin As there was no update on this issue, I'm assuming this is solved with the extra BIOS update for the transition BIOS. I'm closing this issue.

closed 2020-02-02 by texhex