texhex/BiosSledgehammer #93: Not running as expected

GregoryMachin, 2019-11-05:
Hi, 

Please can you advise. 
I'm running the BiosSledgehammer via Commandline in SCCM, under" Command line" Task, this worked ok under WinPE, but under Win7 I'm getting issues where the script is not running fully, this is all it is producing : 

**********************
Windows PowerShell transcript start
Start time: 20191105152336
Username  : WAIKATO\SYSTEM 
Machine	  : HWL63888 (Microsoft Windows NT 6.1.7601 Service Pack 1) 
**********************
**********************
Windows PowerShell transcript end
End time: 20191105152403
**********************
This is the code I'm using to run it :
%windir%\System32\WindowsPowershell\v1.0\Powershell.exe -Executionpolicy bypass -file BiosSledgehammer.ps1 -Verbose

What have I missed ?

texhex, 2019-11-10:
Indeed, this is very strange (the launch command looks good to me). I have no idea what is causing this, except maybe an outdated/broken PowerShell installation maybe.

Can you try to execute the script on the device directly and post a screenshot here please?

texhex, 2020-02-02:
@GregoryMachin I'm closing this issue, it seems to be dormant. Please feel free to reopen if you think the issue is not solved.

closed 2020-02-02 by texhex