texhex/BiosSledgehammer #95: Intel Software Guard Extensions (SGX) changes require key presence

bitslacker, 2020-01-28:
I've run into an issue where changing Intel Software Guard Extensions (SGX) setting requires physical presence - 4 digit key press. Have you been able to get around this requirement?

texhex, 2020-01-30:
No, this can be bypassed, this is hard coded. Any change, that the BIOS sees as an decrease in security (e.g. turning SGX off) will trigger a PPI prompt. The other way around, e.g. activating SGX should go through directly. 

You can turn of PPI, but this change will trigger an PPI prompt itself. Also, I would not recommend this at all, PPI is to prevent "under the hood" changes that might weaken your security settings without anyone noticing. 

If you really, really need to get PPI out of your way, the only way would be to talk to your HP sales person if HP can deliver devices with custom BIOS settings (factory set to your defaults). But I believe they only offer this in some markets and require a minimum amount of devices orders.

bitslacker, 2020-02-01:
I think you meant to say this **can't** be bypassed, right? Alright this makes sense, and I appreciate the quick response!

By any chance do you know if the BIOS can be updated if the Intel Software Guard Extensions (SGX) setting is set to "Software Control" instead of "Enabled"? I will test Monday in the lab but thought I'd ask just in case you're familiar with it.

texhex, 2020-02-02:
Yes, you are right, this was an typo. Sorry. It should read: It can't be bypassed. 

I'm not aware that there is any difference for the BIOS update process if the SGX setting is changed. However, we do not use SGX. 

I could think that if SGX requires some sort of "activation" for a SGX enclave (key enrollment) and you try an BIOS update after this that *also* includes an fTPM update, there might be an additional prompt as most likely the enclave key will be deleted.

closed 2020-06-19 by texhex